prusa3d/PrusaSlicer #2914: Bridge lifting

espr14, 2019-09-13:
v 2.1.0 rc win64

Center of bridge falls down during printing and go a little bit up during cooling. However, they never get to the straight line. Let's build bridges different way.

Instead of going straight while printing bridge line, go up with the nozzle a little bit and then back at the end. The more closer to the center of the bridge, the higher the nozzle should go.

The level of lifting should be defined by user and should be recomputed according to the length of a bridge. Next layer should be put on the bridge after a while (defined by a user) and it should keep its layer height.

Jebtrix, 2019-09-13:
I'd be interested to see results of trying that. Have you did any custom g-code to test that idea?

espr14, 2019-09-14:
Me too. Not yet.

espr14, 2019-12-12:
I've tested it by increasing Z of end of the bridge line. This causes blob when it gets back to a pillar. Maybe it would need extra retraction.

Since the bridge can be printed so slow that the thread is not falling down. The issue is only those few millimeters just at the tip of the nozzle. So the nozzle should lift up (1 or 2 mm) and the place where it leaves pillar and sit back the edge of the pillar on the other side.

closed 2026-09-29 by prusaslicer-bot[bot]