prusa3d/PrusaSlicer #2917: Sliding the layer goes in the opposite direction (OS-X MAC users)

IgnacioHR, 2019-09-13:
### Version

2.1.0-rc2+

### Operating system type + version

OSX - Mac

### 3D printer brand / version + firmware version (if known)

MK3S+MMU2S

### Behavior

Once the piece is laminated. Moving the selected layer using the mouse wheel moves the layer "in the opposite direction".

This might be because you are handing the low level mouse events rather than something already processed by the operating system. For MAC users we "move the orange mark" so when we slide the mouse wheel UP we want it to move UP. Note that windows goes the other way around. In windows you move the "slicer" and the document flows in the opposite direction.

[image]

wabbitguy, 2019-09-18:
Never used Windows but on my Mac's it looks like it's the opposite of whatever System Preferences --> Mouse --> Natural Direction is set at. I prefer Natural Direction unchecked and no issue with the way PRUSASlicer presently works.

Perhaps a preference option is the best route.

bubnikv, 2019-09-18:
PrusaSlicer shall read out this "Natural Direction" flag.

@YuSanka would you please look into it? Thanks.

bubnikv, 2019-09-26:
We spent 3 woman days and 1 man day investigating the issue and iterating the options. We believe the current behavior is natural on OSX, it reflects what the other scroll bars do in the system. The UI toolkit wxWidgets that we are using provides the scroll values depending on the "natural vs. windows" system scrolling configuration. We have a way to detect the "natural" settings and flip the scrolling back on the single scroll bar mentioned, but we are opposed doing so if there is only one person asking for it, and our in house OSX expert tells us the way it works already is the way he expects it to work.

I am leaving this issue for others to voice their opinion. It the majority says the current behavior is wrong, we will flip it. Otherwise we will close this issue in a month.

IgnacioHR, 2019-09-26:
Thank you for the time spend on this. I'm also software developer and I know how this things might go crazy sometimes.

I've found this on wxwidgets web site.

[https://docs.wxwidgets.org/trunk/classwx_mouse_event.html#a189cb1fe40f2590029d8e5c640187574](https://docs.wxwidgets.org/trunk/classwx_mouse_event.html#a189cb1fe40f2590029d8e5c640187574)

Cannot you simply add a GUI option to "honor natural scrolling in vertical sliders" so we as users have the choice to decide?

If you read carefully this thread, at least two Mac users on this post have said the control goes "anti-natural".

bubnikv, 2019-09-26:
IsWheelInverted() is a method that is not supported by the wxWidgets
version that we are using, but @YuSanka back ported it already to our
wxWidgets 3.1.1 fork.

Let's wait for a bit before we collect more feedback.

517nate, 2020-09-03:
I too am a mac user, please please give us an option.

closed 2026-09-29 by prusaslicer-bot[bot]