prusa3d/PrusaSlicer #2919: Supports for areas between surfaces above the build plate

bman2000, 2019-09-14:
2.0

When I use a support blocker to add support for a horizontal slot above the build plate, the support is always generated starting from the build plate.

I only want support inside the slot. 

Even regular support generation does the same thing. The model has no need for support starting at the build plate.

Jebtrix, 2019-09-15:
You mean you used support enforcer? Why would you use blocker to add support? Place an enforcer and use For support enforcers only generation type. If that doesn't work correctly attach a project file for people to examine.

Jebtrix, 2019-09-15:
Actually if you still have issues v2.1.0 is going to be officially released very soon. I wouldn't bother fighting 2.0 problems.

bman2000, 2019-09-15:
Correct. I meant enforcer. I look forward to the new version.

Thank you

Betnie

closed 2019-09-15 by bman2000