prusa3d/PrusaSlicer #2920: Another annoying warning dialog (as #2326)

liftbag, 2019-09-14:
### Version
_PrusaSlicer-2.1.0-rc2_

### Operating system type + version
_macOS Mojave version 10.14.6_

### 3D printer brand / version + firmware version (if known)
_it's a slicer issue_

### Behavior
* _When set 100% cubic infill a warning dialog does not allow to go on_
* _Steps needed to reproduce the problem_
  * _set cubic infill, set 100% density. A warning dialog appears, click OK_
* _Expected Results_
  * _The warning closes, while the filling density value is left as it was_
* _Actual Results_
  * _The warning closes and immediately after it reopens, always the same, preventing other operations_

[image]

#### Project File (.3MF) where problem occurs
_Any, I suppose._

Jebtrix, 2019-09-15:
I can reproduce this also on Linux. If you right click an object and add Infill, you automatically get Fill Density and Fill Pattern object settings in Infill groupbox. If you change to 100% on a sparse infill type, you get warning, click OK and then Fill Pattern is automatically changed to Rectilinear. This is the same behavior as changing that in global settings.

However if you remove Fill Pattern from the Infill groupbox, and then change to 100% this where the logic loop problem starts. Understanding what it wants to do here its easy to see why it breaks. In object settings either the Fill density has to keep old value or Fill pattern has to be automatically added back to object settings to follow how global settings behave.

bubnikv, 2019-09-16:
It is a duplicate of #2878, isn't it?

@YuSanka is working on a fix, but we shall release the 2.1. I feel it is an annoying bug, but there is a workaround, and we will fix it in 2.1 for sure.

bubnikv, 2019-09-16:
This issue will be fixed in PrusaSlicer 2.1.0 final. Closing.

closed 2019-09-16 by bubnikv