prusa3d/PrusaSlicer #2923: Supports for areas between surfaces above the build plate #2919

bman2000, 2019-09-15:
additional information re above issue
[outer.zip](https://github.com/prusa3d/PrusaSlicer/files/3613625/outer.zip)

Jebtrix, 2019-09-15:
You should have just edited your original issue. I understand what your saying. I also agree PrusaSlicer should be able to generate supports this way.

**You expect this:**
[image]

**PrusaSlicer will only do this:**
[image]

bman2000, 2019-09-15:
At least you see the issue. In the meanwhile I use Cura for models like this and Pruss slicer for most others.

Thank you

Ps
Sorry I did not know I could edit a closed issue.

closed 2019-09-15 by bman2000