prusa3d/PrusaSlicer #2924: [Suggestion] Automatically use different infill pattern for 100% infill

polymetric1, 2019-09-16:
I frequently switch between 100% and 20% infill on the same profile, depending on the print. However, switching from 20% grid infill to 100% infill will force the slicer to change to rectilinear infill. I prefer grid for non-solid infill.

I'm suggesting this: "Top fill pattern" and "Bottom fill pattern" be replaced with one option, "Solid infill pattern", and this gets used automatically when infill is 100%.

rongith, 2019-09-16:
I'm sorry to inform you that "Top fill pattern" and "Bottom fill pattern" were split since version 1.42 (I think). And I hope it will stay separate.
You could do 2 derivatives profiles for your needs. But you may already know that and it's not what you would like to do.

Sembazuru, 2019-09-18:
Also, the top and bottom fill patterns are for modifying the external layer fill on the top and the bottom. I've used this for aesthetic effect (some filaments look really neat with Hilbert curve on the bottom layer with a well tuned Live-Z.)

rtyr, 2021-02-03:
I suggest to use two separate profiles if you are frequently switching between 20% and 100% in one session. Closing.

closed 2021-02-03 by rtyr