prusa3d/PrusaSlicer #2927: \Appdata\Roaming\PrusaSlicer\PrusaSlicer.ini corrupted when using Guest Wifi

cmmtchll, 2019-09-16:
### Version
2.0.0

### Operating system type + version
Windows 10 64 bit Enterprise, 1903, 18362.356

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S - FW 3.8.0

### Behavior
Start PrusaSlicer (even tried downloading 2.1 final and same problem).  Immediate crash on start.  Error is: "Internal error: Error parsing PrusaSlicer config file, it is probably corrupted.  Try to manualy delete the file to recover from the error.  Your user profiles will not be affected.

c:\Users\XXXXXXX\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini
<unspecified file>(16): '=' character not found in line"

Contents of the file show the problem:
...
version = 2.0.0+win64
version_check = 1
**version_online = <!DOCTYPE html>
<html lang="en">
    <head>
        <meta http-equiv="refresh" content="0;URL='https://norwegian.cabin.network'" />
    </head>
    <body>

    </body>
</html>**
version_online_seen = 2.0.0
view_mode = expert
...

The problem is that PrusaSlicer checked for the online version of PrusaSlicer when my computer was connected to a Wifi on plane and I was not signed in yet.  This resulted in reading garbage instead of expected content.  Slicer didn't validate and corrupted the file.  This is QUITE serious in that there is no telling what PrusaSlicer could have read...**It needs to validate (presumably with a certificate check) that it is reading correct info.**

_Is this a new feature request?_
No

vojtechkral, 2019-09-16:
Thank you for the report, I'll fix this...

closed 2019-09-20 by bubnikv

turgaycinbat, 2021-11-16:
> ### sürüm
> 2.0.0
> 
> ### İşletim sistemi türü + sürümü
> Windows 10 64 bit Kurumsal, 1903, 18362.356
> 
> ### 3D yazıcı markası / sürümü + ürün yazılımı sürümü (biliniyorsa)
> MK3S + MMU2S - Yazılım 3.8.0
> 
> ### Davranış
> PrusaSlicer'ı başlatın (hatta 2.1 son ve aynı sorunu indirmeyi denediniz). Başlangıçta anında çökme. Hata: "Dahili hata: PrusaSlicer yapılandırma dosyası ayrıştırılırken hata oluştu, muhtemelen bozuk. Hatadan kurtulmak için dosyayı manuel olarak silmeyi deneyin. Kullanıcı profilleriniz etkilenmeyecektir.
> 
> c:\Users\XXXXXXX\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini (16): '=' karakteri satırda bulunamadı"
> 
> Dosyanın içeriği sorunu gösteriyor: ... version = 2.0.0+win64 version_check = 1 **version_online =
> 
> ```
> </body>
> ```
> 
> ** version_online_seen = 2.0.0 view_mode = uzman ...
> Sorun şu ki, bilgisayarım uçakta bir Wifi'ye bağlıyken ve ben henüz oturum açmamışken PrusaSlicer PrusaSlicer'ın çevrimiçi sürümünü kontrol etti. Bu, beklenen içerik yerine çöplerin okunmasıyla sonuçlandı. Dilimleyici dosyayı doğrulamadı ve bozdu. Bu ÇOK ciddi, çünkü PrusaSlicer'ın ne okuyabileceğini söylemek mümkün değil... **Doğru bilgiyi okuduğunu (muhtemelen bir sertifika kontrolü ile) doğrulaması gerekiyor.**
> 
> _Bu yeni bir özellik isteği mi? _ Nu

turgaycinbat, 2021-11-16:
[image]
[image]

turgaycinbat, 2021-11-16:
Without deleting the slicing program, I set my pc back 1 week, restored it, created an installation point, after I restored it, I gave the prusaslicer 2.3.3 installation command again and it was fixed.

commit 8e11a7b8 (2019-09-16): PresetUpdater: Fix: Sanitize downloadedsemver, fix #2927
Use HTTPS for the slicer app version url