prusa3d/PrusaSlicer #2929: only able to run one instance of Prusa slicer at a time

Kgoon1951, 2019-09-16:
2.0.0 Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Mac OSX Operating system type + version 10.14.6 (18G95)
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Prusa Mk3s - 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* It seems like I can only have one instance (one window) of Prusa Slice open at a time. I have a multiple monitor setup and would like to work on more than one project at a time without saving , opening and saving projects. Is this possible currently (I overlooked something) or will it be available in a later release?

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-09-16:
Strange, Win7 and Linux multiple instances work fine. Can you open a terminal and try:

`open -n /Applications/PrusaSlicer.app/`

*Check what PrusaSlicer _xxxxxx.app_ name is actually used, I don't own a Mac to verify path.

Jebtrix, 2019-09-16:
Also try the just released new version v2.1.0

bubnikv, 2019-09-17:
On OSX, only one PrusaSlicer application starts. This is how the old Slic3r always worked on OSX.
We may consider to change it.

Kgoon1951, 2019-09-17:
Thank you for your reply. I thought I was doing something wrong. If you
could add this feature in one of your releases that would be greatly
appreciated.

Extremely please with my Mk3s!! You guys make a great printer and software!

Kgoon1951, 2019-09-18:
Is there a way to export all the slicer setting so you can document how you
setup each print. If not I will definitely request it. Do you know if it
has been requested previously?

Thank you

Kelvin

bubnikv, 2019-09-18:
menu
File -> Save project
File -> Export Configuration
File -> Export Config Bundle

liftbag, 2019-09-23:
> On OSX, only one PrusaSlicer application starts. This is how the old Slic3r always worked on OSX.
> We may consider to change it.

If you set different application names, you can open how many instances you want.
For example, you can rename the PrusaSlicer application to PrusaSlicer 2.1 and then reinstall it.

[image]

As a result, several separate instances of the same application can be opened

[image]

In the meantime that something will be changed, this can be a useful workaround.

Kgoon1951, 2019-09-23:
That’s great! Thanks so much for your help.

bubnikv, 2019-09-23:
cool!

bubnikv, 2020-10-12:
PrusaSlicer will allow opening new instances in the upcoming 2.3-alpha1 from the menu (Window->Open new instance) and from the context menu of the docker icon.

closed 2020-10-12 by bubnikv