prusa3d/PrusaSlicer #2930: Feature Request: support 3.8 sheet profile selection

PRGeno, 2019-09-16:
### Version  2.1

### Operating system type + version  ALL

### 3D printer brand / version + firmware version Prusa Mk3S 3.8 firmware

### Behavior
Now that the 3.8 firmware is out which adds steel sheet profiles, the next obvious Prusa Slicer feature for me would be to support calling those profiles from within the G-Code produced by Prusa Slicer.  I would think it would be a profile selector that allow choosing 1-8 from the Printer profile tab   

_Is this a new feature request?_ YES

bubnikv, 2019-09-17:
That sounds like a risky proposition to me. IMHO one should verify at the
printer that the sheet on the printer matches the printer configuration.

bubnikv, 2019-09-17:
You may try to convince the firmware team https://github.com/prusa3d/prusa-firmware/issues
to implement a G-code for that, and then you can add it to your printer profile start G-code.
Closing here.

closed 2019-09-17 by bubnikv