prusa3d/PrusaSlicer #2932: Height Range Modifier applies to entire model if Start=0

CapnBry, 2019-09-16:
### Version
Version 2.1.0+win64

### Operating system type + version
Windows 10 Pro 64bit

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
When using the new Height Range Modifier (sweet feature BTW), when I add any settings, the settings take effect over the entire object and use **both** the original and the modified settings which leads to infill in perimeters or double infill:
[image]

* Add 40mm cube to bed
* Right click "Height range Modifier"
* Right click the modifier in the list, add a "Layers and Perimeters" or infill
* In the case above I changed Perimeters from 3 to 5, note that now the entire model has 5 perimeters and also has 3 perimeters, both on top of each other and also in the infill.

#### Workaround 1
This can be worked around by setting the "Start at height" to non-zero, e.g. 0.01 or 1 or 10 or whatever so it seems somewhere is doing a `if (start height != 0)` instead of `if (start height >= 0)`. 

#### Workaround 2
This can also be worked around by turning on "Clip multi-part objects" in Print Settings -> Advanced, since it always works if that feature is on (although I do not have a multimaterial printer).

### Project File (.3MF) where problem occurs
[cube4040.zip](https://github.com/prusa3d/PrusaSlicer/files/3618009/cube4040.zip)

Jebtrix, 2019-09-16:
Use Expert mode to see option
Enable Print Settings > Advanced>  Other > Clip multi-part objects

Jebtrix, 2019-09-16:
More details on Clip multi-part objects setting:

Issue [https://github.com/prusa3d/PrusaSlicer/issues/2803](https://github.com/prusa3d/PrusaSlicer/issues/2803)
Issue [https://github.com/prusa3d/PrusaSlicer/issues/2887](https://github.com/prusa3d/PrusaSlicer/issues/2887)

CapnBry, 2019-09-16:
Ah, I knew there had to be an issue open for this but I could not find it. I'll close this one and follow those. Thanks!

"Clip multi-part objects" should not be required to be turned on IMHO, since no other modifier meshes or right-click add modifier requires it to be turned on to generate non-overlapping gcode.

closed 2019-09-16 by CapnBry