prusa3d/PrusaSlicer #2933: Incorrect bottom place planarity

ddavidebor, 2019-09-16:
### Version
2.1 rc2

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macos

### Behavior
When trying to print the attached STL file, the two planar faces are not interpreted as such, instead they are interpreted as follows:

[image]

[ltk-mw_device_fulcrum.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3618386/ltk-mw_device_fulcrum.3mf.zip)

Jebtrix, 2019-09-16:
Couldn't tell from simply looking at mesh but a quick scale Z 0 to the bottom faces in Blender flattened and slices correctly. So it was technically not perfectly flat. 

[ltk-mw_device_fulcrum_flat.zip](https://github.com/prusa3d/PrusaSlicer/files/3618422/ltk-mw_device_fulcrum_flat.zip)

ddavidebor, 2019-09-16:
Yes sorry it appears the whole model was rotated and the STL generated incorrectly, it was an issue with the generating software
@Jebtrix thank you very much for your help

closed 2019-09-16 by ddavidebor