prusa3d/PrusaSlicer #2934: Feature Request: Start Gcode Enhancement

malcinator, 2019-09-17:
### Version
Prusaslicer 2.1.0 Combined with Firmware 3.8.0

### Operating system type + version
macOS 10.14.6 or any

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S
Prusa i3 MK2.5S

### Enhancement
Add these to Prusaslicer config ini files

; Prusaslicer Settings Used
; Printer Settings - [printer_settings_id]
; Filament Settings - [filament_preset]
; Print Settings - [print_settings_id]

I have two Prusa Printers & and couple of other brands, with the arrival of 2.1.0 Prusaslicer & firmware 3.8.0 I see that there is a check for the correct sliced code being printed on the matched printer with M862.3 P "[printer_model]". I use different profiles for almost every type of filament, Print Settings & Custom Printer Profiles and combining this with some simple automation via a mapped share & Octoprint 'Watched folder', the M862 is getting in the way of my workflow, I could for example print a sliced MK2.5S file on a MK3S and vice versa, now I get the M862.3 beeping at me. Currently I can't see a way of finding out what printer & profiles I had used to slicer with, so if I send the wrong file to the wrong printer I have to re-slice it and this adds some time to my workflow.
 
So my question is would it be possible to add something like the above to the Prusaslicer ini files so it would be possible to see in the code which printer & profiles I had selected? 

_Is this a new feature request?_
Yes

bubnikv, 2019-09-17:
This request is better suited to be posted at
https://github.com/prusa3d/PrusaSlicer-settings
for @rtyr to take care of.

malcinator, 2019-09-17:
DOH! Of course, I'll move it over.

JohnnyDeer, 2020-02-10:
@malcinator yes please, if issue is moved, or solved, you can close it.

closed 2020-02-10 by malcinator