prusa3d/PrusaSlicer #2936: prusaslicer 2.1.0 manual filament swap problems

jon-cpu, 2019-09-17:
### Version
2.1.0

### Operating system type + version
Windows 10 pro versjon 1809

### 3D printer brand / version + firmware version (if known)
Stock prusa MK3S firmware 3.8.0

### Behavior
* _Describe the problem_
After prusaslicer 2.1.0 i can no longer follow this tutorial without a purge block being created.
https://manual.slic3r.org/advanced/filament-swaps
Prusaslicer 2.1.0 creates a purge block if extruders is set to 2 in settings, even if printer use profile without mmu.
This new behavour prevents the use of M600 to do a manual toolchange if you dont want to change the filament for a whole layer.
The purgeblock creates alot of gcode, that is difficult to identify and remove manually. 

* _Steps needed to reproduce the problem_
Use a stock mk3s profile without mmu, go to printer settings and set extruders to 2.

_Is this a new feature request?_
Unsure, please either fix to old behavour where the purgeblock is not created if we dont have a mmu, or give option to disable the purge block.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[mod1-drev2-1.zip](https://github.com/prusa3d/PrusaSlicer/files/3620558/mod1-drev2-1.zip)

lukasmatena, 2019-09-17:
> (There is a error in this tutorial: you cant set the extruders to 1 again

The manual you refer to is for upstream Slic3r, not for PrusaSlicer.

> give option to disable the purge block

It is in there for years (Print Settings -> Multiple Extruders -> Wipe Tower)

jon-cpu, 2019-09-17:
PrusaSlicer is a fork of Slic3r if i understand correctly, so atleast some things seems to be correct still :)
Also it seems Prusaslicer does not have any manual?

Thank you very much for the location of the wipe tower settings, worked a charm :)

closed 2019-09-17 by jon-cpu

lukasmatena, 2019-09-17:
> Also it seems Prusaslicer does not have any manual?

You are right. We diverged too far from the upstream and its manual is no longer usable. The manual for current PrusaSlicer is being made and will be available soon.