prusa3d/PrusaSlicer #2937: [Feature Request] Custom bed model offset

AmyTheCute, 2019-09-17:
### Version
V2.1

### Operating system type + version
Windows 10 64-bit

### 3D printer brand / version + firmware version (if known)
Hypercube Evolution with Klipper

### Behavior

I load my Custom bed model into the bed section.
The corner of my bed appears in the middle.

if there was an offset i could align it precisely with the 0,0 of my bed.

_Is this a new feature request?_ Yes!

#### Project File (.3MF) where problem occurs
N/A

Jebtrix, 2019-09-17:
I would just check and adjust the origin of the mesh model itself in the mean time.

liftbag, 2019-09-18:
> I would just check and adjust the origin of the mesh model itself in the mean time.
This is exactly what I do.

[image]

AmyTheCute, 2019-09-25:
How do you change the model origin ! Mines an stl from thingiverse so i dont have any cad files for it

Jebtrix, 2019-09-25:
You don't need CAD file, STL is fine. Knowing some [Blender](https://www.blender.org/download/)is always a good skill. This should be enough [Blender 2.8 QUICK TIP to Move The Object Origin and Pivot](https://www.youtube.com/watch?v=q_SiM8PIXtI). If you still have trouble just provide the thingiverse link to this model.

rtyr, 2021-02-03:
Sorry, I don't think this is a task for PrusaSlicer. The proper way is to fix the stl. Closing.

closed 2021-02-03 by rtyr