prusa3d/PrusaSlicer #2938: Shortcuts needed for  Mirror Object and Top / Bottom Layer

prowlmedia, 2019-09-17:
**Shortcuts needed for** 
Mirror Object X / Y / Z ( currently hidden in a 2 step right click > Mirror > Along X Axis )  
Top / Bottom Layer when sliced

Suggest
**Mirror x  = Shift + X 
Mirror y = Shift + Y
Mirror z = Shift + Z
Top Layer = Shift + Up Arrow
Bottom Layer = Shift + Down Arrow**

bubnikv, 2019-09-17:
>  Top / Bottom Layer when sliced

*> Top Layer = Shift + Up Arrow> Bottom Layer = Shift + Down Arrow*

What do you mean with that?

prowlmedia, 2019-09-17:
In the layer window... so you can jump viewing from the top layer to the bottom layer - often I only need to check the first layer is correct and then the top layer.  So to be able to just instantly form top to bottom or vice verse would be very helpful

prowlmedia, 2019-09-17:
Great will check that out - It’s not mentioned anywhere... Not if you could add the mirror keystrokes that would be awesome.

prowlmedia, 2019-09-19:
Ok - I have checked out the left right arrow as suggested... The cursor jumps but not the visual representation of the slices...

I just want to press a keystroke to jump to View the First layer on the bed to make sure it's correct and then the last layer to see the final built model.  Instead of having to drag the slider down.

neophyl, 2019-09-19:
If all you want o do is look at the bottom later on the bed and the top of your model why not just use the numeric keypad keys 1 and 2 ?  1 views the bed from above and 2 the bed from below.  (0 shows the 3/4 on view and other numbers show other angles) .

prowlmedia, 2019-09-19:
> If all you want o do is look at the bottom later on the bed and the top of your model why not just use the numeric keypad keys 1 and 2 ? 1 views the bed from above and 2 the bed from below. (0 shows the 3/4 on view and other numbers show other angles) .

Good interim solution - 2 does indeed show the bottom of the sliced model so I can see the bottom layer... but won't show just the one layer without doing the slide of the top dome control.

hoegge, 2020-03-22:
I'd like to advocate against shortcut keys for mirror. Mirror is something you rarely do, and you can risk  ruining a print by accidentally mirroring an object

prowlmedia, 2020-03-22:
so YOU rarely mirror things. I do it EVERY single print.   Left and right handed products.  The keyboard shortcut can be difficult. ALT+SHIFTX/Y/Z and some might say thats your fault for not checking it before export. I do. 

Or CTRL+Z undo it! 

I’d actually say that a proper keyboard shortcut system you could edit would be better so they actually matched whatever 3d app each person uses. 

Annoyingly this actually existed in Mac as you can change ANY Mac menu text keyboard shortcut in preferences but the devs of this have removed most of the standard Slic3r too menus for some unexplained reason.  And Mirror was in there.

hoegge, 2020-03-22:
@prowlmedia Not really sure what you mean in 

> The keyboard shortcut can be difficult. ALT+SHIFTX/Y/Z and some might say thats your fault for not checking it before export. I do.

Are there any keyboard shortcuts now? What is my fault? Not checking if it was mirrored? You can say that, but my issue is I don't have a clue how it got mirrored, since I did for sure not do it using the menu.

prowlmedia, 2020-03-22:
No there are no shortcuts at all for mirroring at the moment! So no idea how you did it The only way is if you make the local scale -100 on an axis.  Or drag the scale gizmo. 

I mean if you make a mistake that’s the users   Fault that they didn’t check it.  Happens to me sometimes when I forget to check the bottom layer and I know that’s my fault.  Like it would be if you mirrored it and don’t check. 

I mean I don’t mind if the keyboard shortcut are more difficult than Say a single key... Like X but can  have shift+X  Or shift+ctrl+X

hoegge, 2020-03-22:
@prowlmedia Strange - I never mirror or scale and only print STLs from CAD where I do all the modifications. Only rotate and place them. 
I agree keyboard shortcuts are nice for things you do often - but since mirroring is dangerous, I'd then prefer as you suggest SHIFT+X/Y/Z or ALT+SHIFT+X/Y/Z so they don't by mistake are hit when people try to hit copy / paste keys and don't notice a quick mirroring (there is probably not a CTRL+X function in PrusaSlicer, but still :-)

prowlmedia, 2020-03-22:
If nobody needed to Mirror... why is the function there at all :)  Because people requested it in Slic3R ages ago. 

Mirroring is only dangerous if you don’t check your slice before outputing.  I could say the same about scale. 

If you create as many objects as I do that require mirroring... it’s much easier to have 1 STL export And mirror them than say 100%_X,  -100%_X and the other permutations.

hoegge, 2020-03-23:
Who is saying nobody needs to mirror? 

Yeah - and driving is only dangerous if you crash. Come on. I say mirroring that is easy to do by mistake (single keystroke) is 'unforntunate' since people might not see it happen and why would they look for something, if they never do it? You should not have to - and cannot - check your slice for everything - you check for the things you modify (often) and you know of. 

I'm not at all advocating against mirroring anywhere in my posts. I'm just saying, it should be a shortcut you don't hit easily by mistake - IF there should be shortcuts for that at all.

prowlmedia, 2020-03-23:
arghhh - You did :) 
"Mirror is something you rarely do" 
"I never mirror or scale"

Single keystroke?   I never said that... and there isn't one now!? My original post said...

Mirror x = Shift + X
Mirror y = Shift + Y
Mirror z = Shift + Z
Top Layer = Shift + Up Arrow
Bottom Layer = Shift + Down Arrow

"You should not have to - and cannot - check your slice for everything - you check for the things you modify (often) and you know of."

Yeah you should! to see if it's floating / not sitting flat.  
That the Slicer has actually worked correctly
That everything is stable and won't collapse - that perimeters are ok etc

I have no remorse for anyone who doesn't check they are exporting something that takes hours to print. That is insane... or if they are doing it for bulk CNC they would be fired!

Do you just hit GCODE then print?!

mneumark, 2020-08-08:
I'd just like to note that as a Mac user, sometimes when I can't find a feature (like this time), I go to the Help menu and use the search box. It will search for text in all the menu and then highlight stuff it finds. But if its hidden behind a right click option it won't show up there. I highly recommend something as simple as mirror (and other similar things like creating objects) be in a menu at the top.

closed 2026-09-29 by prusaslicer-bot[bot]