prusa3d/PrusaSlicer #2940: Very poor bridging logic?

SnorreSelmer, 2019-09-17:
### Version
2.1.0 Release

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Tronxy X5S-400 w/ factory firmware

### Behavior
* _Describe the problem_
I'm trying to print a costume helmet, and the top will not be printable, because for some reason, Prusa Slicer decides to not support it from the bed up, but wants to bridge a massive gap and build the final supports from there
* _Steps needed to reproduce the problem_
I can't share the .STL or .3MF file here, but I could email it to someone at Prusa. (It's a paid model, and I've signed an NDA, so I hope people understand.)
* _Expected Results_
The flat part on the top of the model to be supported from the bed up.
* _Actual Results_
[image]

Jebtrix, 2019-09-17:
It should be enough to export just the Config and attach that to rule out configuration.

neophyl, 2019-09-17:
Personally Id turn off bridge detection, then add a support enforcer under that area.

As a matter of interest though I recently printed off a  140 x 120mm cylinder with a flat top and I was amazed that my printer actually managed to bridge the long spans (modified direct drive CR10S with a microswiss hotend, 0.6 nozzle and an upgraded hot end fan.  So its not always an impossible bridge :)

Prusa Slicer could do with a user definable 'max bridge length' though.  I think others have requested something along those lines previously.

Jebtrix, 2019-09-17:
I approximated the model with a dome and flat top since I didn't see your model top and I have no problems either way.

[image]

SnorreSelmer, 2019-09-18:
I managed to solve it by doing as neophyl suggested.. I looked around a bit
for bridging-related settings and disabled them one by one until I solved
my issue. And I agree with neophyl that a max bridging distance setting
could be useful here. I've printed test-bits to find bridging- and
overhang-limits, so I know how far I can push it before I get in trouble.

closed 2026-09-29 by prusaslicer-bot[bot]