prusa3d/PrusaSlicer #2941: Crash while slicing due to out of RAM

daktylus, 2019-09-17:
### Version: 2.1.0+win32
Build: PrusaSlicer-2.1.0+win32-201909160915

###Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18362), 64-bit edition

WorkingSet: 763MB
PrivateBytes: 647MB
Pagefile(peak): 647MB(1,126MB)
RAM size reserved for the Undo / Redo stack: 422MB
RAM size occupied by the Undo / Redo stack: 0MB
OpenGL installation
GL version: 4.6.0 NVIDIA 388.13
Vendor: NVIDIA Corporation

### 3D printer brand / version + firmware version (if known)
Prusa MK2S

### Behavior
Slicing crashes due to too low RAM even more RAM is available. (total RAM size 16 GB)
Model slices without problem with 0.15 mm layer heigt (17 hours print time). It crashes while slicing it with 0.10 mm layer height

[image]

#### Project File (.3MF) where problem occurs
Sorry I cannot provide the stl file

Jebtrix, 2019-09-17:
Your using a win32 version on a win64 OS. Try the PrusaSlicer win64 version.

daktylus, 2019-09-18:
OMG you are right. I clicked the wrong donwload link. That's so stupid. Thanks !

closed 2019-09-18 by daktylus

bubnikv, 2019-09-18:
I suppose we shall report Win32 application running on 64bit windows.