prusa3d/PrusaSlicer #2942: Bed Visually Disappeared, Printer not displaying timer

ChiefGyk, 2019-09-17:
### Version
_2.1.0 Linux x64
### Operating system type + version
Pop!_OS (Ubuntu flavor) 18.04
Version:   2.1.0+linux64
Build:     PrusaSlicer-2.1.0+linux64-201909160925

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 4.15.0-62-generic x86_64
Total RAM size [MB]: 33,611MB
OpenGL installation
GL version:   4.6.0 NVIDIA 435.21
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1050 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
[... 322 lines cut ...]
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

### 3D printer brand / version + firmware version (if known)
MK3S mostly stock, firmware compiled with G81 tweak for the Nyloc mod.

#Issue 
All of a sudden the print bed visual is not appearing in my slicer anymore when all I have done is update it from 2.0 to 2.1, and even now 2.0 has the same issue. I also notice the Slicer no longer adds the time to my output gcode in the file name any more and I know that's not a firmware issue as I had 3.8.0 installed a few days ago and had zero issues until the latest file sliced with Prusa Slicer 2.1.0

But my printer does not display a countdown timer anymore nor a USB progress when working over OctoPrint, 

Here are the attached images
[image]

[image]

Jebtrix, 2019-09-17:
Try renaming ~/.PrusaSlicer profile folder and restart program. Sounds like profile corruption. Not sure on the Octoprint issue.

ChiefGyk, 2019-09-17:
Already tried the renaming of the profile same result. Even on PS 2.0

Jebtrix, 2019-09-17:
Just for clarification you using distro repo or appimage build?

ChiefGyk, 2019-09-18:
I am using it from the Tar.bz2 extracted into a folder, the appimage has never really rendered G-Code properly for me. Now I have a weird issue when it is rendering G-code it says it is made for a different type of printer? But it's literally the code included with the printer except adding the G81 command to the start G-code

Jebtrix, 2019-09-18:
So you still have the original problems with a new one? Can you save a problem project, zip it up and attach it?

ChiefGyk, 2019-09-18:
[Zartman_litho_flat.zip](https://github.com/prusa3d/PrusaSlicer/files/3625810/Zartman_litho_flat.zip)
Attached is a project I am working on with an STL and the Gcode it output. 

I am using the settings I open sourced in my Repo https://github.com/ChiefGyk/Prusa_Settings but there isn't anything wild besides having it do a G81 on start G-code for the printer.

Jebtrix, 2019-09-18:
It would be best if you saved a 3MF project file from PrusaSlicer and zipped that. Your 3D bed model is missing because you need to change your printer ini profiles:

From
`inherits = `
To
`inherits = Original Prusa i3 MK3S`

ChiefGyk, 2019-09-19:
Yup that did it

closed 2019-12-06 by ChiefGyk