prusa3d/PrusaSlicer #2943: Shortcuts to change view direction switch away from isometric view

zbrozek, 2019-09-18:
### Version
2.1.0+win64

### Operating system type + version
Windows 10 Professional

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Viewport defaults to perspective. This can be switched to isometric by pressing '0', but is reverted when using the other shortcuts to switch camera angles.

"Iso" shouldn't be a view, it should be a toggle.

Jebtrix, 2019-09-18:
Iso is indeed a view. I'm guessing you mean Perspective versus Orthographic which can be changed in Preferences. Or are you referring to how some programs automatically switch to orthographic for Top/Bottom/Left/Right and then Iso view as perspective or orthographic depending on global setting?

zbrozek, 2019-09-18:
Oops, you're right - I've mixed up terms. Thank you for setting me straight!

closed 2019-09-18 by zbrozek