prusa3d/PrusaSlicer #2944: Cumulative time for color sliced object is backwards

Sembazuru, 2019-09-18:
### Version
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 17,116MB
OpenGL installation
GL version:   4.6.0 NVIDIA 411.63
Vendor:       NVIDIA Corporation
Renderer:     Quadro K2100M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:
[...345 lines of extensions snipped...]

### 3D printer brand / version + firmware version (if known)
Not important, but Original Prusa i3 MK3S fw:3.8.0

### Behavior
* _Describe the problem_
The accumulated time listed for a multi-color print (non MMU) is backwards.

* _Steps needed to reproduce the problem_
Add more than one color change in the preview layer slider.

* _Expected Results_
The time for the first color printed should be added to the second color printed, then added to the third color printed, etc.

* _Actual Results_
The time for the last color printed is added to the penultimate color printed, etc.
Screeshot of a 4 color print:
[image]

_Is this a new feature request?_
This is a followup to the added feature from https://github.com/prusa3d/PrusaSlicer/issues/2390
Sorry I missed this during the alphas and betas.

bubnikv, 2019-09-18:
> [image: 2019-09-17 23_15_01-DishwasherCatMagnet - PrusaSlicer-2 1
> 0+win64-201909160915 based on Slic3r]

I am afraid it is a matter of taste.

bubnikv, 2019-09-19:
This behavior is by design, it slipped under my radar. Our lead tester asked for it. The 2nd number matches the remaining time number shown by our MK3 printer. This makes sense, but it is certainly not clearly presented.

Sembazuru, 2019-09-19:
Oh, that makes sense (matching the countdown display on the printer). I was just confused. I'll close this now under PEKAC. ;-)

closed 2019-09-19 by Sembazuru