prusa3d/PrusaSlicer #2945: Cannot set Z speed independantly.

DarkAlchy, 2019-09-18:
Version is all of them because in S3D I can set Z axis movement speed but in Prusa Slicer Z is tied to travel moves so treats Z moves the same as X and Y.  This dates back to Slic3r itself and was fine back then but Marlin 2.0 no longer caps the speed you set in the configuration.h.  They are changing that thanks to we who begged them as they were opposed to it.  Anyway for now people with Marlin 2.0 will slice with your slicer and have the travel moves at 130mm/s and Z will do some ugly nasty things as it did to me.

woodencase01, 2019-09-21:
I second that. I had a lot of trouble searching for a clue with Z skipping, after realizing that the gcode generated from Prusa Slicer called a F12000 on the Z axis...

liftbag, 2019-09-23:
I second that too. I'm not sure if the limits set in the firmware of my machine can mitigate an excessive speed set by the gcode.
Recently I redesigned and built the entire extruder of my Cartesian, making it much lighter. This allows me to set high XY travel speeds. But the print bed is always the same, and I doubt it can move at the speed of the extruder's XY axes.
In our telegram group where we discuss prusaslicers, we all noticed a bit better performance of the walls with S3D than in PrusaSlicer. At this point, I wonder if this is due to the fact that S3D separates the travel speeds between XY and Z axes.
Diversifying the XY and Z travel speeds seems like a great idea.

szaghi, 2019-09-23:
I agree, it is really helpful to be able to set Z speed freely from XY one.

adenbley, 2020-04-18:
I'm having the same problem with an Ender5. Is there any workaround other than writing a post-processing script or manually editing the gcode every time?

DarkAlchy, 2020-04-19:
> 
> 
> I'm having the same problem with an Ender5. Is there any workaround other than writing a post-processing script or manually editing the gcode every time?

Nope, so I insert my own custom gcode to handle this issue.

Depending on which slicer you use determines how it is written but the end result is as such:

M201 X1500 Y1500 Z25 E200
M203 X245 Y245 Z3 E100 
M204 P1500 R1500 T1500

lordwimsey, 2020-10-27:
Hello,

I have the same issue, missing steps when my z axis tries to follow the too fast feedrate.
May I kindly asked, whether this issue is being worked on and if there might be a fix anytime soon? 

Thanks in advance!

DarkAlchy, 2020-10-27:
> 
> 
> Hello,
> 
> I have the same issue, missing steps when my z axis tries to follow the too fast feedrate.
> May I kindly asked, whether this issue is being worked on and if there might be a fix anytime soon?
> 
> Thanks in advance!

Doubtful since there has been no signs, or a proper response, to this plus it can be set by you in the header gcode as I laid out above.  A bit sad but to be expected out of Josef.

lordwimsey, 2020-10-27:
> Doubtful since there has been no signs, or a proper response, to this plus it can be set by you in the header gcode as I laid out above. A bit sad but to be expected out of Josef.

Thanks for your reply!
I doubt it will help, since M503 show me values even lower than yours:

> Recv: echo:Maximum feedrates (units/s):
> Recv: echo:  M203 X150.00 Y150.00 Z6.00 E120.00
> Recv: echo:Maximum Acceleration (units/s2):
> Recv: echo:  M201 X2000 Y2000 Z300 E10000
> Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
> Recv: echo:  M204 P1500.00 R1500.00 T1500.00

DarkAlchy, 2020-10-27:
Yes, this is what I said you need to set them in the user defined section of the gcode and make it send them each time.

ipa64, 2021-03-28:
Hi, nothing news about that ?

martinbudden, 2021-04-16:
I've made a PR to address this issue, see https://github.com/prusa3d/PrusaSlicer/pull/6369

DarkAlchy, 2021-04-16:
Excellent.

martinbudden, 2021-04-20:
In the meantime, here's a regular expression you can use to edit the gcode: replace `(\nG1 Z[\d\.]+ +)F[\d\.]+(.*\n)` with `$1F720$2`
This will set the feed rate for the Z axis moves to 720.

lukasmatena, 2021-05-31:
Implemented in master for 2.4.0 release (413bce49fc9b07adc5ee5869c8acf6713c30ba33). Closing.

closed 2021-05-31 by lukasmatena