prusa3d/PrusaSlicer #2946: No "Slice Now" option whe changing Height Range Modifiers

jweaver100, 2019-09-18:
### Version
Version: 2.1.0+win64

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa Mk3  Firmware 3.8

### Behavior
The Height Range Modifier is a nice feature, but the GUI has a few quirks.

Lets say you have a model which is 100mm tall and Slicer is set to a Layer Height of 0.2mm

You then a Height Range Modifer with the start at 30mm and snd at 50mm and set the Layer height to 0.3mm, it works fine

But then say you change the Start or Stop value (because you want to change the area you are working with).. Or the Layer height... You then have no "Slice Now" option, to see the results (the button only shows "Export G-code")

Instead you have to change something else (i.e move the part) just so Slic3r will see that something has changed and allows you to slice.

Also.. The Layer Height option seems a bit odd.. Sometimes you type a number and it just resets to the original value.. For example now in my test.. I can type 0.1 or 0.2 and hit return and they stick.. But if I type 0.3, it goes back to 0.1 when I hit return

There may be some logic to this, but I can't work it out...

FidelCapo, 2019-09-18:
Thank you for report. You are right. PrusaSlicer should display Slice Now if you change these values. You can use "File - ReSlice now (Ctrl+R)" as workaround.
JIRA - 1025 created

For layer height value issue. This is because your "Printer Settings - Extruder - Layer height limits - Max" value is probably smaller then 0.3. But there is another small bug - this behaves differently if you press enter and if leave the textbox.
JIRA - 1026 created.

rtyr, 2021-02-03:
Fixed. Closing.

closed 2021-02-03 by rtyr