prusa3d/PrusaSlicer #2949: Modifier changes extruder (MMU)

s4andst0rm, 2019-09-19:
### Version
2.1.0+win64 (final)

### Behavior
* _Describe the problem_
When applying an (infill) modifier to an object which has parts assigned to different extruders in the modified region, the extruder (thus the color too) becomes fixed for the whole layer.
Since there is no way to assign an infill modifier to just one part of a multiple-part-object I dont see any way to work around it.
Is this a known issue?

Before modifier:
[image]

Modifier settings:
[image]

After modifier:
[image]

#### Project File (.3MF) where problem occurs
3MF of example attached.
[infill_modifier_extruder.zip](https://github.com/prusa3d/PrusaSlicer/files/3630090/infill_modifier_extruder.zip)

lukasmatena, 2019-09-19:
> Since there is no way to assign an infill modifier to just one part of a multiple-part-object

The simplest way to do it is to place the modifier so that it only overlaps the one part:
[image]

But I understand what you're saying - we'll look into it some day.

bubnikv, 2019-09-19:
This is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2657

bubnikv, 2019-09-19:
Closing this one.

closed 2019-09-19 by bubnikv