prusa3d/PrusaSlicer #2950: Start G-Code inserted after setting the temperatures.

daaz68, 2019-09-19:
### Version
PrusaSlicer 2.1.0+win64 

### Operating system type + version
Windows 7 Pro 64bits

### 3D printer brand / version + firmware version (if known)
Custom made delta

### Behavior
I use a custom g-code (M52) to power up the main supply, therefore I entered this in the Custom G-Code/Start G-Code section of the Printer Setting configuration tab. Because this command starts the power, I expected it to be inserted in the output g-code file as the first command.

My "Start G-Code section" looks like this:
    **M52 ; start power source
    G28 ; home all axes
    G1 Z5 F5000 ; lift nozzle**

The g-code in the output file looks like this (temperature settings are configured before the start code):
    M107 ; disable fan
    M190 S100 ; set bed temperature and wait for it to be reached
    M104 S250 ; set temperature
    **M52 ; start power source
    G28 ; home all axes
    G1 Z5 F5000 ; lift nozzle**
    M109 S250 ; set temperature and wait for it to be reached
    G21 ; set units to millimeters

Thank you,
Daniel

lukasmatena, 2019-09-19:
You may want to read the tooltip on Custom G-Code section:

> If PrusaSlicer detects M104 or M190 in your custom codes, such commands will not be prepended automatically.

Does this solve your problem?

daaz68, 2019-09-19:
No, it does not. 
I removed the M104 and M190 instructions from my custom code sections completely, but the generated code still places my code afterwards, as before. Code remains the same in output.

lukasmatena, 2019-09-19:
Insert something as

M104 S[first_layer_temperature]
M190 S[first_layer_bed_temperature]

into your custom gcode. PrusaSlicer will understand that you want to set the temperatures yourself and will not emit any gcode to set it (neither before nor after the start gcode).

daaz68, 2019-09-19:
This (partially) works. The new code looks like this:
M107 ; disable fan
M52 ; start power source
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M104 S250
M190 S100
...
As you can see, M107 still comes first.

Your suggestion implies that I will have to change the temperatures by hand every time I change the filament from ABS to PET or whatever. Not pretty.

A better way to address this would be to run a script after the code was generated and re-arrange the G-code instructions. Still really ugly, but at least the filament temperatures are all in a single location.

lukasmatena, 2019-09-19:
> Your suggestion implies that I will have to change the temperatures by hand every time I change the filament from ABS to PET or whatever.

No, it doesn't. When I suggested to insert "M104 S[first_layer_temperature]" into the custom gcode, I really meant to insert it verbatim, including the square brackets and the expression inside. This will be substituted by the respective value from the Filament config. See also https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

> M107 still comes before anything else

In order to supress the M107 at the beginning, go to Filament Settings->Cooling->Fan Settings and set 'Disable fan for the first 0 layers'. This will prevent generation of that command.

daaz68, 2019-09-19:
Ok, I got it working as intended.

Million thanks for the help, I wouldn't have done it without you.

Yours,
Daniel

lukasmatena, 2019-09-19:
Glad to hear that. Closing the issue.

closed 2019-09-19 by lukasmatena