prusa3d/PrusaSlicer #2953: [minor] 2.1.0 "Set number of instances" triggers Assertion fail Error

mavelli, 2019-09-19:
PrusaSlicer 2.1.0 

* right click on object
* Set number of instances
* OK

triggers error message
An assertion failed! 
/usr/include/wx-3.0/wx/strvararg.h(456): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

but the object gets multiplied correctly anyways. so only minor bug. 

```
Backtrace:
ASSERT INFO:
/usr/include/wx-3.0/wx/strvararg.h(456): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

BACKTRACE:
[1] wxArgNormalizer<long>::wxArgNormalizer(long, wxFormatString const*, unsigned int)
[2] Slic3r::GUI::Plater::set_number_of_copies()
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[9] wxMenuBase::SendEvent(int, int)
[10] g_closure_invoke
[11] g_signal_emit_valist
[12] g_signal_emit
[13] gtk_widget_activate
[14] gtk_menu_shell_activate_item
[15] g_closure_invoke
[16] g_signal_emit_valist
[17] g_signal_emit
[18] gtk_propagate_event
[19] gtk_main_do_event
[20] g_main_context_dispatch
[21] g_main_context_iteration
[22] gtk_main_iteration
[23] wxWindow::DoPopupMenu(wxMenu*, int, int)
[24] wxWindowBase::PopupMenu(wxMenu*, int, int)
[25] Slic3r::GUI::Plater::priv::on_right_click(Slic3r::GUI::ArrayEvent<Eigen::Matrix<double, 2, 1, 2, 2, 1>, 1ul>&)
[26] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[27] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[28] wxEvtHandler::TryHereOnly(wxEvent&)
[29] wxEvtHandler::ProcessEventLocally(wxEvent&)
[30] wxEvtHandler::ProcessEvent(wxEvent&)
[31] wxEvtHandler::ProcessPendingEvents()
[32] wxAppConsoleBase::ProcessPendingEvents()
[33] wxApp::DoIdle()
[34] g_main_context_dispatch
[35] g_main_loop_run
[36] gtk_main
[37] wxGUIEventLoop::DoRun()
[38] wxEventLoopBase::Run()
[39] wxAppConsoleBase::MainLoop()
[40] wxEntry(int&, wchar_t**)
[41] Slic3r::CLI::run(int, char**)
[42] main
[43] __libc_start_main
[44] _start
```

Archlinux x64 - Package versions:
prusa-slicer 2.1.0-1
boost-libs 1.69.0-2
curl 7.66.0-1
glew 2.1.0-1
intel-tbb 2019.8-1
nlopt 2.6.1-1
wxgtk2 3.0.4-2
qhull 2019.1-1

lukasmatena, 2019-09-20:
You're apparently linking against wxWidgets 3.0. The assert does not trigger when linked against 3.1, which is what we do with our builds. I suggest you do the same. It is already quite hard to make PrusaSlicer work on all three major platforms with one wxWidgets version, I don't think someone will have time to look into this soon.

https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

mavelli, 2019-09-20:
https://bugs.archlinux.org/task/63862

lukasmatena, 2019-09-20:
Ok, I at least looked into where the assert bubbled from and this is apparently really a mistake in our code. I haven't tested, but it will be most likely solved by https://github.com/prusa3d/PrusaSlicer/commit/656569b0e9e836c628f85cabc16e6805382ac729

But the reasons for linking against wxWidgets 3.1 still hold. There are workarounds in our code that probably won't work in versions different from 3.1 that we tune them on.

bubnikv, 2019-12-17:
The assert was fixed. Closing.

closed 2019-12-17 by bubnikv