prusa3d/PrusaSlicer #2954: [FR] Import brim "ears" from Slic3r

pscrespo, 2019-09-19:
### Version
2.1.0

_Is this a new feature request?_
Yes.

Is it possible to add brim "ear" feature like the one present in Slic3r? It would help with some ("pointy"? :sweat_smile: ) 3D models.

_More info:_ 
Issue: https://github.com/slic3r/Slic3r/issues/4687
PR: https://github.com/slic3r/Slic3r/pull/4691

[image]

Jebtrix, 2019-09-19:
Current workaround using modifiers: [#2680](https://github.com/prusa3d/PrusaSlicer/issues/2680)

[image]

imro2, 2020-05-01:
> 
> 
> Current workaround using modifiers: [#2680](https://github.com/prusa3d/PrusaSlicer/issues/2680)
> 
I just spent hours banging my head, trying to reproduce this, but was not having any luck. Turns out, this workaround does not work if the Elephant foot compensation under Print Settings -> advanced is not set to 0. Hope that helps somebody else.

I hope the merge request will be implemented soon, so we don't have to monkey with workarounds.

dattrax, 2020-11-28:
Seems like the SuperSlicer fork has this feature.  Any chance we can get it into PrusaSlicer?

bubnikv, 2020-11-30:
> Seems like the SuperSlicer fork has this feature. Any chance we can get it into PrusaSlicer?

One day.

sarvenn, 2021-01-11:
I tried prusal slicer rc3 but I still can not see brim tab(ear) function which is available in Slic3r and Cura. I wish new v2.3 prusaslicer had it either.

4wrxb, 2021-04-26:
This was pretty high in my google search so updating the suggestion from here:

> Current workaround using modifiers: [#2680](https://github.com/prusa3d/PrusaSlicer/issues/2680)
> 
> [image]

If you create an object as mentioned the slicer in 2.3.1 gives an error that an object has no extrusions on the first layer. I have posted a work-around in the issue there as it does not seem to be demoted to a warning https://github.com/prusa3d/PrusaSlicer/issues/4062

imro2, 2021-04-27:
I tested it the old way and even though it shows the error, it seems to slice ok.

[image]

imro2, 2021-04-28:
Turns out that you can work around this limitation by adding a modifier and overriding the no bottom/top layers and perimeters. As long as it prints a single dot, it is going to pass the test. So the steps to get this working:

1. make sure elephant foot compensation is turned off. (Print Settings -> Advanced -> Elephant foot compensation - set to 0)
2. load your stl
3. Right click on the build plate and Add Shape (box or cylinder depending on your needs)
4. select the newly added object and under Object manipulation click the pad lock to disable proportional scaling
5. set the z size to the height of your first layer
6. move the object in place.
7. Right click the newly created object and select Layers and Perimeters, this will add Object Settings to Modify to your right pane.
8. Set Bottom, Perimeters, Top to 0
9. Right click the object again and select Infill
10. Set Fill density to 0   <--------- Up to here are the old steps
11. Right click the object again and select Add Modifier -> Cylinder
12. Scale down the Cylinder to 1 x 1 x 1 and move it into place. Make sure it overlaps with at leas a part of your add object
13. Right click the modifier and select Layers and Perimeters
14. Set the Perimeters in Part Settings to Modify / Layers and Perimeters in the right pane to 1

In the example below, the black box is your stl. Orange is the added object set to first layer height and positioned. Green is the modifier for your added object.
[image]

If you need more than one of these, just select the object, CTRL+C, CTRL+P

4wrxb, 2021-04-30:
Your notes provide a similar set-up (with more detailed instructions). It sounds like the modifier cylinder works "inside" the keep-out object where the grouped object had to be outside the volume to work so that's easier to deal with.

sanchosk, 2021-06-16:
>     12. Scale down the Cylinder to 1 x 1 x 1 and move it into place. Make sure it overlaps with at leas a part of your add object

Make sure that after the scale down, the modifier SITS ON THE BUILD PLATE! Mine was, for some reason, few mm up, so even though I added the modifier, there was no real intersection with the added 1-layer cylinder. Moving it down manually and voila - I have my fake elephant ears :)
Still, it would be great to have the elephant ears functionality similar to Cura...

sanchosk, 2021-06-17:
Is there a way how I can sponsor adding this feature to PrusaSlicer? I believe it would be very beneficial for people without heated bed and, in some cases, also for people with heated bed but with difficult models.

Eldenroot, 2022-12-17:
Bump this... I have to use Cura for this particular feature (only for specific models)... quite annoying. Please add it into Pusa slicer (or at least port it from Slicer)

redPanther, 2023-02-20:
> I just spent hours banging my head, trying to reproduce this, but was not having any luck. Turns out, this workaround does not work if the Elephant foot compensation under Print Settings -> advanced [...]

this feels not like a workaround. it feels like it is working due to a bug. Very anoying and I understand brim is low prio, but since superslicer implemented this and both slicer had same/similar base it would be nice if a developer could take some days to port this into prusaslicer.

currently I want to have a textured finish on the first layer (will be to of my model) and therefore I need to use the textured bed. Unfortunately PLA doesn't adhere so good. I use 3DLAC and it work quite well, but thin corners need brim support.

lorenzofanchi, 2023-07-01:
+1 Would be great to see this in PrusaSlicer. One of the reasons I still user SuperSlicer these days.

wikolii, 2024-01-23:
Please, pretty please add this to PS

tk-notify, 2024-05-20:
Would be great to have the brim ear option. The mentioned work arounds are not working for me. Looking forward to an implementation.
Thanks!

pscrespo, 2024-08-20:
Hi! I've seen on duplicated issues that SuperSlicer implemented this feature, and I found the following related commits:

https://github.com/supermerill/SuperSlicer/commit/b1b76c07235dbe55ea6f350c17e1e8143fb08643
https://github.com/supermerill/SuperSlicer/commit/3ec2f077e5e43bfc21e18b3faac9211685731bfb
https://github.com/supermerill/SuperSlicer/commit/d4fe1e573d76dfd5d8befe7e60a928624e7f6812
https://github.com/supermerill/SuperSlicer/commit/1ed31fc4b80059bc75dff5066442bf32e1fa14c3

Hope this helps. Sadly I'm not a programmer so I can't help implementing it.

Snuff1eupagus, 2024-08-31:
Maybe this will help some of you to achieve your goals with Brims

Selective Blocking
https://www.youtube.com/watch?v=6CRqNk_JUj4

Selective Adding
https://www.youtube.com/watch?v=dhsbMmosbLU&t=1s

Regards

slavkosky, 2025-01-23:
Still waiting for a feature like this. I tried using a custom cylinder shape I made 3.5 x 3.5 x 0.2mm and imported. I add the shape as an additive modifier, but it wants to print with a single perimeter on first/top layer (my project requires this, so this is enabled project-wide). I tried overriding those settings on just the modifier shapes, however disabling them does nothing to the sliced result. It seems there is a bug where these parameters (and a couple others I've noticed over time) aren't respected by the slicer _if they are set on a modifier object_ 

[image]

imro2, 2025-02-18:
@slavkosky you were almost there. Not sure what you did differently but similar steps worked for me. Came here to post only to find out you beat me to it.

1. right-click the the object you need to add ears to in the right pane and select Add part -> Cylinder. This should create a new group with your object and the newly added cylinder in it.
2. change the height of the cylinder to the first layer height, scale the x/y size, and place it where you need it
3. right-click the cylinder in the right pane and and select Layers and Perimeters
4. set perimeters to something else that what your object has. I used zero perimeters
5. order the cylinder so that it is above your object in the list within the group
6. optionally CTRL+C - CTRL+V to duplicate the cylinder and place it around as needed

Note: the reason the perimeter of the object is not straight is the elephant foot compensation. if you set it to 0 it will be straight.
[image]

redPanther, 2025-02-19:
> [@slavkosky](https://github.com/slavkosky) you were almost there. Not sure what you did differently but similar steps worked for me. Came here to post only to find out you beat me to it.
>
>     1. [...]

as previously stated out, what a user unfriendly thing! Nice that we have a workaround, but why is this so much clicks! could be simple as right click -> add part -> brim ear.
- this creates a cylinder that has automatically the height of first layer
- is always ordered before main object (aka printed before main object)
- also perimeter is set to 0 for this object

you can xy scale it and place it to your needs with existing tools. I could imaginge a rectengular shape to for doing partial brim on straight edges

Snuff1eupagus, 2025-02-20:
I guess the links were broken, I can't fix laziness, but at least you can know how.

Selective Adding

https://www.youtube.com/watch?v=0Zxl1WOJm90

Selective Removal 

https://www.youtube.com/watch?v=IEKUpZ-gjMs&t=24s

Regards

imro2, 2025-02-20:
> I guess the links were broken, I can't fix laziness, but at least you can know how.
> 
> 
> Selective Adding
> 
> https://www.youtube.com/watch?v=0Zxl1WOJm90
> 
> 
> 
> Selective Removal 
> 
> https://www.youtube.com/watch?v=IEKUpZ-gjMs&t=24s
> 
> 
> Regards

With all due respect and kudos for discovering and documenting what seems like an unintended, albeit very useful side effect, calling people lazy doesn't seem productive.

As previously stated, all of these are workarounds that utilize some sort of Prusa slicer quirk that could change or go completely away with new release.

All that would be required is a new object that already possesses all the settings you talk about in your video - negative volume, inner brim, first layer height, etc.

anghelos, 2025-06-12:
I just want to point out that Prusa's Easyprint has a mouse ears feature.

It adds tiny 0.6x0.6x0.1mm cylinders close to the corners, and adds a brim to those.
In order for the brim to properly contour your actual object, it sets global brim width to 0.01mm, which doesn't actually create a brim, but still includes your main object to the calculation.

[image]

Snuff1eupagus, 2025-06-21:
@imro2 

Freaking losers, I can't even pretend anymore.
Oh BTW, I know you won't know this but Prusa is using the most simple form of this because they weren't show by Swiss_Cheese how to use the more advanced form, on purpose, just to see how this would turn out.

The community is crap, this is why Swiss isn't making videos anymore. The community had years to share and learn these things and all they do is act helpless, "OH GOD Make A button for this" otherwise its a work around, you deserve this.

closed 2026-09-29 by prusaslicer-bot[bot]