prusa3d/PrusaSlicer #2957: Confusing terminology "Perimeters" versus "Shells"

bobstro, 2019-09-20:
### Version
2.1.0+ (and older Slic3r)

### Operating system type + version
MacOS 10.14.6 on 5K display

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3

### Behavior
Under **Print Settings->Layers and perimeters->Vertical shells**, the terms "shells" and "perimeters" are used to describe the same thing. This is very confusing when trying to explain this feature to someone. Is there a distinction? Is there a reason different terms are used within 10mm of screen space?

Jebtrix, 2019-09-20:
Huh? If we were talking about a building and I wanted options for the Wall it would be how many Bricks thick to make it. I don't see the confusion.

bobstro, 2019-09-20:
Wouldn't have disagreed until I found myself trying to explain this all to someone. The fact that gap fill (a 3rd undefined term!) is used without a specific on/off setting makes this particularly confusing. These are all sliced with 2 perimeters specified:

[image]

Try explaining to someone designing a part how thick they should make their walls and you'll see where it gets to be very cumbersome to explain. "Your wall is a shell which is made of of an external perimeter, zero or more non-external perimeters that you specify (even though the tool tip says your setting is a minimum), plus maybe some gap fill or sparse infill, depending on some hidden parameter." I had to create and slice parts to have any chance of explaining what was going on. Having gap fill hidden under speed settings doesn't help, but I've opened another issue on that.

In the absence of any real documentation, wouldn't simply using "wall" make more sense? "Shell" implies the outer-most layer, which is easily confused with "external perimeter". Of course, a good set of docs would help alleviate the problem. I realize this goes back to the Slic3r base, but this might be a consideration for future usability and UI improvements. I'm pretty good at technical documentation, but explaining this process gave me a headache.

Jebtrix, 2019-09-20:
Devs mentioned real documentation is going to be made by a team of their content people. Shell is the standard terminology for every slicer I've seen. Cura throws in the concept of Skin along with Walls under Shell category, Simplify3D settings use "Outline/Perimeter Shells". Everybody has their flavor of terminology, though different none are inaccurate.

bobstro, 2019-09-21:
True, but I don't think there's an existing slicer that wins awards in terms of clarity, intuitiveness or good UI principles. I'm just pointing out that the mixing of terms is less than clear, and the results less than obvious. I understand it now, but certainly not from the names or tool tip descriptions. Not exactly a "feature request" but a suggestion for future consideration.

bubnikv, 2019-09-23:
I passed the link to our content guys, who are currently working on slicer manual. They also take care of localization, they tend to harmonize our UI texts, maybe they will feel inspired.

closed 2026-09-29 by prusaslicer-bot[bot]