prusa3d/PrusaSlicer #2959: Crashed during slicing

vmzsoft, 2019-09-20:
[image]
[image]
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/3635649/config.zip)
[STL.zip](https://github.com/prusa3d/PrusaSlicer/files/3635651/STL.zip)
But any *.stl...
And always this error:
[image]

YuSanka, 2019-09-20:
Thanks for your report. But we can't reproduce this issue. Could your please click the "Show problem details" button. And send us a screenshot.

lukasmatena, 2019-09-20:
Thanks for the report. I can confirm there is something wrong, I also saw it crash, although it does not happen everytime on my system (Ubuntu 16.04). When run in debug mode, there is an assert failing:

LayerRegion.cpp:133: `assert(surface.surface_type == stInternal)`
Actual content of surface.surface_type is stInternalSolid.

I enclose another 3MF to reproduce it (your config differs a lot from our profiles, this might be easier to evaluate).
After playing with it for a bit, it seems to me that the culprit is Print Settings->Infill->Solid Infill Threshold Area. When set to zero, the assert no longer fails. Could you please test whether it is indeed so?

[internal_infill_assert.zip](https://github.com/prusa3d/PrusaSlicer/files/3636350/internal_infill_assert.zip)

vmzsoft, 2019-09-20:
[image]
Error present when slaysing...
Also I see many artefacts with your model:
[image]
Maybe error due to the video card?
I have Mobile Intel(R) 4 Series Express Chipset Family video on my notepad.

lukasmatena, 2019-09-20:
>I load your file and go to slicing. The same error present when slise...

I know. The 3MF I provided is a minimal scenario to reproduce the problem.

Please, set SOLID INFILL THRESHOLD AREA to zero and check whether that solves the problem.

vmzsoft, 2019-09-20:
Ok.
[image]
Then I have the same error
[image]

Jebtrix, 2019-09-20:
I'm not sure if you just reused the same picture but that last Crash screenshot is not showing "internal_infill_assert.3mf(layer_region_crash5.3mf(..." as chosen in the Print Settings combobox.

vmzsoft, 2019-09-20:
O, sorry...
Real screenshot
[image]

Jebtrix, 2019-09-21:
Intel integrated graphic drivers are known to be problematic and that's where it is faulting.  Since you are running Windows 32bit OS I also wonder how much system ram has been allocated as dedicated vram. Things I would try:
* Update video drivers to latest: https://downloadcenter.intel.com/product/80939/Graphics-Drivers
* Run PrusaSlicer without any other programs running (ie Chrome the ram hog)
* I see your using Comodo- temporarily disable HIPS protection (if enabled)
* Increase dedicated video ram in your BIOS (if available, try maximum it allows for any difference). 
* Run PrusaSlicer in software render mode: prusa-slicer-console.exe --sw-renderer

That screenshot you attached with render artifacts is a little troubling.  Since you are using integrated gpu and its using system ram there is a chance you may have ram going bad. Usually this results in a black screen on bootup but it depends where the bad address range is. Being a laptop you can also have thermally induced intermittent issues. You can try running [Memtest86](https://www.memtest86.com/downloads/memtest86-4.3.7-iso.zip) to test your ram as a sanity check and/or [Prime95](https://www.mersenne.org/download/) for a stress test.

vmzsoft, 2019-09-22:
Wow!!!
I use only one piece of advice:
prusa-slicer-console.exe --sw-renderer
and all fine!

[image]

No errors.
No artefacts.
No need to set SOLID INFILL THRESHOLD AREA to zero.
Good slicing for all models.
Brilliant!

bubnikv, 2019-09-23:
There may or may not be a problem with your OpenGL drivers. It may very
well be, that the use of software renderer just masked some invalid memory
access issue as the memory layout will be somehow different with the
software renderer. @lukasmatena was able to reproduce the initial issue, we
will look into it.

lukasmatena, 2023-11-21:
The problem with solid infill area threshold should be fixed by now. Closing.

closed 2023-11-21 by lukasmatena