prusa3d/PrusaSlicer #2961: M109 not recognized in custom G-code

svucko, 2019-09-20:
### Version
Version 2.1.0+linux64

### Operating system type + version
$ uname -a
Linux LIFEBOOK-E751 4.15.0-64-generic #73-Ubuntu SMP Thu Sep 12 13:16:13
### 3D printer brand / version + firmware version (if known)

### Behavior
* M109 not recognized in custom G code and prepended with automatic M109
* Add "M109 S[first_layer_temperature]" in fillament Custom G-code
* _Expected Results_
No automatic M109 added before custom G-code

Result:
G1 Z15 F5000 ; lift nozzle
G1 E2 F800 ; prime extruder
M73 P1 R3
;END START G-CODE
M109 S245 ; set temperature and wait for it to be reached   <----- automatically added
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
;Purge line start
G90 ;absolute positioning
G21 ; set units to millimeters 
G0 X3 Y4 F9000 ; Go to front
G0 Z0.2 ; Drop to bed
G92 E0 ; zero the extruded length
M109 S245 ; set extruder final temp   <---- not recognised
G0 E7 F300; Purge Bubble
G1 X60.0 E12.5 F1000.0 ; intro line
G1 X100.0 E17.5 F1000.0 ; intro line
G0 Y5 F9000 ; Go to front
G1 X60.0 E20 F1000.0 ; line reverse
G92 E0.0
;Purge line stop 
G1 Z0.200 F18000.000
G1 E-2.30000 F12000.00000

#### Project File (.3MF) where problem occurs
[Cube_2.1_bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3637372/Cube_2.1_bug.3mf.zip)

bubnikv, 2019-09-23:
The M109 code is searched for in the custom start G-code only, it is not
searched for in the filament start G-code. You need to move your M109 code
to the custom start G-code.
In the start G-code you may address the filament temperature using the
macros.
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

svucko, 2019-09-25:
version PrusaSlicer-2.0.0-rc+linux64 does reckognise M109 in fillament custo G-code.
Exact same configuration is working correctly in 2.0.0 version.

Jebtrix, 2019-09-25:
> Exact same configuration is working correctly in 2.0.0 version.

Both version 2.0.0 final and the 2.0.0 rc version you mention still required a M104 or M109 in Printer Start G-Code to be detected. I quick tested against both. Your project generates the same gcode with automatic M109 insertion working as intended. How you are getting a different result is beyond me. 

In the end it doesn't matter because its how it works now. What is so problematic about putting a M104 or M109 in Printer Start G-Code? 
* Add M104 S[first_layer_temperature] to end of Printer Start G-Code so it starts heating but continues. Your custom M109 gcode will be processed a moment later anyway.
* Add M104 S0 to do nothing but suppress auto M109 insertion. It will act the same exact way you had before.
* Use this opportunity to speed up your time to print and add M104 S\<_choose temp under melting point_\> to the beginning of your Printer Start G-Code. Let it start heating while it homes and gets ready. That way no chance of ooze at lower preheat temp target but your M109 will not take as long now.

Jebtrix, 2019-09-25:
BTW, is there is reason you have `G1 E2 F800 ; prime extruder` before extruder temp is even set yet? Cold extrusions usually just grind filament into a nice powder.

rtyr, 2020-05-22:
Behavior was explained, so I believe we can close this.

closed 2020-05-22 by rtyr