prusa3d/PrusaSlicer #2962: [Optimization] Bed with single part toggling printable forces reslice

Jebtrix, 2019-09-20:
### Version
v2.1.0

### Operating system type + version
Any

### Behavior
Toggling Printable with a single part on the bed shouldn't require a re-slice. I mainly did this by accident but it was on a couple minute slice time part.

Jebtrix, 2019-09-20:
Or cant toggle printable with only one part restriction maybe?

bubnikv, 2019-09-23:
I don't quite understand. Enabling / disabling an object will cancel G-code
export.

Jebtrix, 2019-09-23:
It's mainly user error but I've done it more than once to myself hence the request. I'll try to explain better.

**Scenario: One object on bed that just happens to take a little while to slice**
* I inspect the result in Preview but it needs tweaking
* I accidentally click Printable toggle when I was trying to right click on Editing column to add values I want to review
* Previous g-code generation is invalidated, Preview is now empty
* Click Printable and hit Slice Now because I need to see it in Preview again to plan out tweaks
* Have to wait for lengthy reslice to see the same g-code Preview again

This is a minor but this is one scenario I wish PrusaSlicer could save me from myself :D

Jebtrix, 2019-09-23:
Of course you could do this accidentally with multiple objects but there is no way of knowing if it was intended. A single object however, whats the point of making your only object non-printable?

bubnikv, 2019-09-24:
I understand it may be annoying. We could add a confirmation dialog if the
slicing is running. This is something to think over.

closed 2026-09-29 by prusaslicer-bot[bot]