prusa3d/PrusaSlicer #2964: Incorrect output extension when dot in filename

Salamandar, 2019-09-21:
Current behaviour :

* input file: `my_file_v1.2.stl`
* output file: `myfile_v1.2`

Correct behaviour : 
* output file: `myfile_v1.2.gcode`

The extension is not correctly determined if a dot is in the filename.

Salamandar, 2019-09-21:
I guess the function `boost::filesystem::change_extension` does not do the job correctly here : 
https://github.com/prusa3d/PrusaSlicer/blob/c1cb40c4f08262b6f7a538d8ea87829e5b9fcc72/src/libslic3r/PrintBase.cpp#L68

Salamandar, 2019-09-21:
I discovered the issue was reported here : #1918 but it's not fixed. I'm using PrusaSlicer 2.1.

Jebtrix, 2019-09-22:
I can't reproduce that problem. Default Prusa profiles and filename scheme output fine. If I use the example as given in that issue #1918 formatted as:
`{input_filename_base}_FD{filament_diameter[0]}_EM{extrusion_multiplier[0]}_LH{layer_height}.gcode `
It also outputs fine.

bubnikv, 2019-09-23:
You have to put the .gcode extension at the end of your output file name
template in case the template may produce a dot.
I know it is a workaround, but it will work.

Salamandar, 2019-09-25:
@bubnikv OK. But workarounds are not fixes ;)

Jebtrix, 2019-09-25:
I actually appreciate having the file extension defined like that. I have a program that wants an .nc extension. Saves me a manual filename change after the fact.

Salamandar, 2019-09-25:
Sure, but users are used to have .gcode "by default" :)

Salamandar, 2026-09-24:
I'll see when 3.0 is available for testing on linux.

Salamandar, 2026-09-24:
Looks like it's fixed.

closed 2026-09-24 by Salamandar