prusa3d/PrusaSlicer #2965: [Feature Request] Need access to XY size compensation in object editing

woodencase01, 2019-09-21:
### Version
Version 2.1.0 Win64

### Operating system type + version
OS Name	Microsoft Windows 10 Pro
Version	10.0.16299 Build 16299

### Behavior
I'm trying to print a shell with some local reinforcements (thicker regions with infill).
I have two parts where one is the outer shell and the other is the reinforcement.
I can configure the perimeters and infill settings per part to get something that is very close to what I need.
[image]

However, one perimeter from the reinforcement is overlapping, check below when set to infill without perimeters:
[image]

By having access to XY Size Compensation in the object editing, I could set the reinforcement to "-extrusion_width" and have the overlapping problem gone.

Fixing the 3D file would be a lot of work and I'd greatly appreciate having it solved with the slicer option.

I personally think that every option should be made available in the object editing.

Jebtrix, 2019-09-21:
Do you have Print Settings > Advanced > Other > Clip multi-part objects turned on?

woodencase01, 2019-09-21:
> Do you have Print Settings > Advanced > Other > Clip multi-part objects turned on?

Thanks for pointing that out. I tried it, and the result isn't working:
[image]

Since both the shell and the housing contain the reinforced zone, the reinforcement is clipped by the shell.

I really need to have the perimeters including the reinforced region, thus explaining why my shell has the reinforced geometry included. 

If I use the "change type" option to "modifier", I have both parts separated by the shell perimeter, which generate a weakness in the part by not having a perimeter joining the reinforced section and the shell:
[image]

Jebtrix, 2019-09-21:
It's a little hard to help without the project file. Did some similar type tests of what your trying to do and I don't get overlapping so I'm not sure what you have going on in your project.

woodencase01, 2019-09-21:
[Shell Example.zip](https://github.com/prusa3d/PrusaSlicer/files/3639006/Shell.Example.zip)

Sorry, should have added this file since the first topic!

Looking forward to your solution :)

Jebtrix, 2019-09-21:
[image]

[Shell.Example_tweaked.zip](https://github.com/prusa3d/PrusaSlicer/files/3639082/Shell.Example_tweaked.zip)

woodencase01, 2019-09-21:
I get the same result. I'm pretty sure the lines overlap on the external shape. Is there a way to check each line one by one with PrusaSlicer?

Jebtrix, 2019-09-21:
Not sure how thats possible.  

Pull bottom control up to Select single layer:
[image]

**Pics from layer 24 as example**

Slice from tweaked project:
[image]

Slice from project you attached (changed nothing):
[image]

Slice from project you attached, changed Perimeter to **1** to exhibit you overlap example :
[image]

Jebtrix, 2019-09-21:
Gcode looks fine. There is the projects settings of 25% infill/perim overlap in effect here.

[image]

If you want zero overlap that you can see visually from Preview then you could do this:

[image]

Otherwise I'm not sure what you want  ¯\\_(ツ)_/¯

woodencase01, 2019-09-22:
Thanks for your time Jebtrix. I really appreciate.

If there's nothing we can do, I'll try to impose a small interference in the 3D file and use the "modifier" type as my first post, then there should be a slightly better bonding between the two parts. Your overlap settings made me thinking.

I just thought this could be a feature added by the developing team in order to make it possible.

rtyr, 2021-02-03:
Implemented. Closing.

closed 2021-02-03 by rtyr