prusa3d/PrusaSlicer #2966: Remove parameter validation warning

woodencase01, 2019-09-21:
### Version
Version 2.1.0 Win64

### Operating system type + version
OS Name Microsoft Windows 10 Pro
Version 10.0.16299 Build 16299

### Behavior
We are using nozzles sizes from 0.20mm up to 5.00mm with our machines.
When the line width is set to a higher value than 1.00mm, there is the following warning:
[image]
When testing configuration with bigger nozzles, there is about 7 parameters to change and 7 warning to confirm.

It would be very nice if:

- This warning wouldn't appear in expert mode
- This warning would have a checkmark "Do not notify me again"
- This warning would be linked with the variable "nozzle_diameter", the same was as it checks when it's too large with the nozzle. A value of 2 is obviously not 2%. :
[image]

UnfoldAntwerp, 2021-11-05:
Same here, printing with ceramic printer with nozzles up to 3mm. A setting in preferences to disable these warnings could also be an option.

rtyr, 2022-01-19:
The threshold for the warning message was increased to 50mm. Closing.

closed 2022-01-19 by rtyr