prusa3d/PrusaSlicer #2967: Please add "Add color change after perimeters"

daktylus, 2019-09-21:
### Version
PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa MK2s

### Feature Request
I want to highlight some text in the first layer of my print by using a different filament color. I wish to add a color change after the perimeters in the (first) layer are finished. Most likely quite easy for you to implement but very useful.

Maybe with a small pop-up menu after clicking the "+", as shown in the image

[image]

lukasmatena, 2019-11-07:
I think that there are more similar cases where one might want to put the toolchange. Maybe it would be best to first do an in-layer preview (so that a slider would show the extrusions one by one, not the whole layer at a time) and only then extend colorprint to add the toolchanging gcode at current position.

What you need to do should be possible by a workaround using simple custom gcode magic. I added an extruder in Printer Settings, turned off wipe tower, pretended that I have a multimaterial printer and set the perimeters on top layers to the second extruder and than changed custom toolchange gcode to include the M600 command. I haven't tested but I assume the printer would skip an automatically added Tn command and let you do the toolchange through the M600. I'm enclosing a 3MF to illustrate it.
[text_highlight.zip](https://github.com/prusa3d/PrusaSlicer/files/3819719/text_highlight.zip)

daktylus, 2019-11-22:
Let me show you could do with it. https://www.thingiverse.com/make:711254

Did this with manual filament change

jhartlep, 2020-04-16:
This would be a great feature!

janfw95, 2020-05-11:
I agree. I missed that feature for some time as well! Would be so great if that would come with the next update. :)

daktylus, 2020-10-07:
Some more results as inspiration ;-)

[image]
[image]

hawwwran, 2024-05-07:
I just got this exact idea, that I want to change the color after perimeters are done to "wipe" the new color into the infill, to prevent imperfections in perimeter. I knew I wasn't the first. It would be nice this task would get more attention.

Like this:

[image]

closed 2026-09-29 by prusaslicer-bot[bot]