prusa3d/PrusaSlicer #2968: Strange extra travel movements after toolchange generated

jmd73, 2019-09-21:
### Version
PrusaSlicer 2.1.0 (Version 2.0.0 does not generate the extra travel movement)

### Operating system type + version
Windows 10 Pro 64bit (1903)
### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S MMU2S (the selected printer setting in PrusaSlicer)

### Behavior
PrusaSlicer generates extra travel moves at tool change. Looks almost like a purge line.
I have a failed print because of this move (when using 0.30mm draft setting). I can attach a picture of it later if needed.
At layer 7 (0.65mm with 0.15mm QUALITY MK3) there is one extra move. If I change Print Settings (only default settings) the moves change. 0.20mm settings gives two longer lines, and 0.30mm setting gives three lines.

* _Steps needed to reproduce the problem_
Start PrusaSlicer. Add at least a four color object (I have only tried with my object. If I only load three parts, the moves are not generated. (3mf attached below). When it is sliced (I have background processing active), change to layer view and activate "Travel" toolpaths to see the moves.
* _Expected Results_
No extra moves.
* _Actual Results_
Extra moves inserted.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
0.15mm Quality setting:
[image]

0.30mm DRAFT setting:
[image]

#### Project File (.3MF) where problem occurs
Remove the .zip after filename.

[ExtraMoves.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3638819/ExtraMoves.3mf.zip)
_

bubnikv, 2019-09-23:
Please recompile the 2.0.0 final. We moved the 2.0.0 label due to this
issue.

closed 2019-09-23 by bubnikv

jmd73, 2019-12-04:
I can confirm that it is fixed with commit  4ba7cb7 in master branch.

lukasmatena, 2019-12-05:
@jmd73 Thanks for letting us know.