prusa3d/PrusaSlicer #2970: LANGUAGE environment variable not respected

Grawp, 2019-09-21:
### Version
2.1.0+linux64 (.tar.bz2 release)

### Operating system type + version
Gentoo 17.1, Plasma without systemd

### Behavior
#### Environment
`LANG` environment variable set to `sk_SK.UTF-8`
`LANGUAGE` environment variable set to `en_US`
#### Expected behavior
PrusaSlicer starts in **English** language.
#### Actual behavior
PrusaSlicer starts in **Czech** language.

### Note
I suspect that `LANGUAGE` is being ignored and `LANG` being `sk_SK.UTF-8` makes the app start in Czech language.

Every other application I can think of correctly use `LANGUAGE` for determining the language to use. I tested setting `LANGUAGE` to `cs_CZ`, `sk_SK`, `en_US` and running some applications (Firefox, Dolphin, ...) and it is really is adhered to in spite of `LANG` being constantly leaved to `sk_SK.UTF-8`.

Jebtrix, 2019-09-21:
There is no specific sk locale translation in PrusaSlicer:
https://github.com/prusa3d/PrusaSlicer/tree/master/resources/localization

It defaults to en_US when it can't find a match. There is a cs_CZ czech locale.

Grawp, 2019-09-21:
@Jebtrix It seems you haven't read my first post at all.
I do NOT want Slovak language. I do NOT want Czech language. I DO want **English**.
I do get English in every other application but the PrusaSlicer which starts in Czech.

Jebtrix, 2019-09-21:
I've read it. I'm sure every other application you run isn't using wxwidgets either.

Grawp, 2019-09-21:
> There is no specific sk locale translation in PrusaSlicer

It should not even look for `sk` when the `LANGUAGE=en_US`

> It defaults to en_US when it can't find a match.

It does not in my case. It starts with `cz`!

> I'm sure every other application you run isn't using wxwidgets either.

True. If it is wxwidgets problem I guess I'll report it upstream.

Jebtrix, 2019-09-21:
`rm -rf /usr/share/PrusaSlicer/localization/cs/ `

Jebtrix, 2019-09-21:
Just noticed you used a tar ball but you get the gist.

Grawp, 2019-09-21:
I can also just set `LANG=en_US` for this one application as a workaround.

I like to have `LANG` set `sk_SK.UTF-8` just to have EUR as default currency, 24-hour clock format, yyyy-mm-dd date format etc.. and `LANGUAGE` set to `en_US` so I have everything in English language. It does work. I really have everything in English language while keeping nice date-format etc..

It just doesn't work with PrusaSlicer 2.1.0. And it worked with 2.0.0.

Jebtrix, 2019-09-21:
Yeah I remember a commit where they did a manual match override for cz and sk, saying they would understand czech so its better than no translation.

[Commit: Slovaks understand Czech well. Give them the Czech translation. ](https://github.com/prusa3d/PrusaSlicer/commit/002ae9a4bfbb302364bbf10bbdebfdb58e6d8435)

Jebtrix, 2019-09-21:
> I can also just set LANG=en_US for this one application as a workaround.

I haven't had the displeasure of wrestling languages and locales on my PCs but that seems more reasonable if it works for wxwidgets detection :)

Grawp, 2019-09-21:
> [Commit: Slovaks understand Czech well. Give them the Czech translation. ](https://github.com/prusa3d/PrusaSlicer/commit/002ae9a4bfbb302364bbf10bbdebfdb58e6d8435)

Yeah. That is totally OK. Actually it's very good idea!

So before that commit it must still had been looking for `sk` just the fallback is now different.

The question then remains whether `LANGUAGE` environment variable should be considered and whether it should have priority over `LANG`. As I said .. it seems to work that way for other applications.
Since it is as you says related to wxwidgets I guess we can close this now.

Jebtrix, 2019-09-21:
Looking at Posix standards your `LANGUAGE` should override `LANG`

Grawp, 2019-09-21:
So it really is a bug (even if it is in wxwidgets) :)

Jebtrix, 2019-09-21:
Of course some distros bend Posix rules so double check. I doubt this would be an area to be non Posix compliant.

bubnikv, 2019-09-23:
Dear Michal.

This is on purpose. The Czech and Slovak languages are pretty close, and
our fathers and grand fathers understand both languages very well. They may
understand some Russian, but they may have issues with English. It is for
our fathers and grand fathers (and mothers and grand mothers), that the
application starts in Czech for Slovak languages.

I believe it will be no issue to you to go to menu Settings -> Select
language (you will surely understand the Czech spelling of it) and to
switch to English. The application will remember your decision.

Vojtech

bubnikv, 2019-09-23:
Also we don't currently have enough resources to localise to Slovak
ourselves, but you are most welcome to contribute.
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/Localization_guide.md

bubnikv, 2020-03-20:
Works as intended. Slovaks get Czech dictionaries by default.

Grawp, 2020-03-20:
I do apologize for not replying sooner. I didn't have time to properly reply back then in september then forgot about it.

@bubnikv You completely misunderstood the issue. @Jebtrix got it. It is NOT about Slovak language at all.
```
LANG="de_DE.UTF-8"
LANGUAGE="en_US"
```
environment should yield PrusaSlicer in **English** language and NOT in **German** language.
At least that's how most Linux applications work.
Even
```
LANG="cs_CZ.UTF-8"
LANGUAGE="en_US"
```
should yield **English** language IMHO.

@bubnikv I agree with you on the matter that localizing into Slovak would be an utter waste of time. I used it ONLY as an example of the problem. (Btw. I've actually been living in Brno for more than a decade now and even before then Czech language had never been a problem for me.)

It really is useful to have language switched to English so you can talk with global community in the same terms, share screenshots etc.. while it is also useful to have LANG switched to cs_CZ/de_DE/sk_SK to have European date and time format Kc/€ as default currency instead of $.. etc.

Manually switching language to English also does not work because it is trying to switch to British English and there is no other English in the dialog to switch to.
[image]

One can always edit `PrusaSlicer.ini` so it is a low priority issue I guess.

bubnikv, 2020-03-20:
Me reading the github issues is like a practice in speed reading. There is so much streaming in and I should be programming instead.

I see the issue now. Navigating localization swamp of Windows / OSX / Linux is quite time consuming indeed.

The British English is the preference hard coded in wxWidgets as the authors are Europeans I supose.

bubnikv, 2020-03-26:
This issue is hopefully fixed with a1aee69c5a501706a5ff3f98e2ed87742224a08b

There is a convoluted code in wxWidgets, which at the end of the day reads LANGUAGE. I had no idea what it does on Unix, so this path was disabled on Unix systems. After reading the wxWidgets source code of wxTranslations::Get()->GetBestTranslation() and friends, I believe it should do now what you guys expect it to do.

The fix will be released with the next release, which may take time. You may test if you compile the master from source code.

Closing.

closed 2020-03-26 by bubnikv