prusa3d/PrusaSlicer #2972: When adding new instance, the object is unselected

ridercz, 2019-09-22:
### Version

2.1.0+win64

### Operating system type + version

Windows 10

### Behavior

When adding new instance, the object is unselected. Therefore if user wants to add more instances, he has to re-select the object.

#### Steps to reproduce

1. Select object
2. Add new instance by clicking `(+)` icon or pressing `Num+` on keyboard.

*Actual behavior:* New instance is added and object is deselected, so repeating step 2 won't add another instance.
*Expected behavior:* Object should remain selected so repeating step 2 would add new instance(s).

enricoturri1966, 2019-09-23:
@ridercz
I am unable to reproduce this bug with 2.1.0+win64
I get the expected behavior:
1) Select object
2) Add new instance by clicking (+) icon or by pressing [+] or [Num+] on keyboard
3) A new instance is added to the scene and it is selected
Now repeat 2) as many time as needed, and I get all the new instances as expected, with latest one selected

liftbag, 2019-09-23:
> @ridercz
> I am unable to reproduce this bug with 2.1.0+win64
> I get the expected behavior:
>
> 1. Select object
> 2. Add new instance by clicking (+) icon or by pressing [+] or [Num+] on keyboard
> 3. A new [...]

I'm unable too with 2.1.0+macOS 10.14.6.

ridercz, 2019-09-23:
Here is video recording of the screen. Compare with decreasing number of objects, which works as it should. https://youtu.be/t0M-Qc3_QLg

ridercz, 2019-09-23:
I just found out that the problem occurs only when auto-center parts setting is enabled and updated original report accordingly. @liftbag  and @enricoturri1966, can you please verify it with centering enabled?

enricoturri1966, 2019-09-24:
@ridercz 
Thank you, now I can repro

enricoturri1966, 2019-09-24:
Working on it -> https://dev.prusa3d.com/browse/SPE-1031

liftbag, 2019-09-24:
@ridercz I can repro too with auto-center parts enabled.

FidelCapo, 2019-09-24:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2019-09-24 by FidelCapo