prusa3d/PrusaSlicer #2973: Crash when using arrange button

u01pei, 2019-09-22:
### Version
2.1.0+

### Operating system type + version
MacOS Mojave v10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa MK3s, completely stock, firmware version 3.8.0

### Behavior
printing the markerbod T-rex [https://www.thingiverse.com/thing:275091](url) opened files "Tail A_C" and "Tail B_D". I used the split into objects on both files to get 5 loose objects. Then deleted 2 of those objects and pressed the "arrange" button in Slicer. The program immediately crashes. I tried several times, but the program always crashes.
Tried the same steps with only file "Tail_B_D" (3 objects), deleted one, pressed arrange, works ok. Then repeated steps with only file "Tail_A_C" (2 object), deleted one, pressed arrange and program crashes again. tried again, but now deleted the other object, program crashes again.

#### Project File (.3MF) where problem occurs
Project saved at the point where I would press the arrange button and experience a crash.
[TRex_Tail_Split.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3639540/TRex_Tail_Split.3mf.zip)
[config.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/3639547/config.ini.zip)

FidelCapo, 2019-09-23:
Thank you for report. I can reproduce your issue.
JIRA 1028 created.

closed 2019-09-24 by FidelCapo

u01pei, 2019-09-24:
Great! Thanks and keep up the good work!