prusa3d/PrusaSlicer #2974: STL model not beeing auto-repaired on import; slicing is incomplete.

cristiansen-12, 2019-09-22:
**### Version**
PrusaSlicer-2.1.0+linux64-201909160925

**### Operating system type + version**
Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.2.11-200.fc30.x86_64 x86_64
Total RAM size [MB]: 33,687MB

**### 3D printer brand / version + firmware version**
Original Prusa i3 MK3s -- stock
Firmware Version: 3.8.0-2684

**### Behavior**
* _Describe the problem_
Importing this particular STL model is successful. No errors or warnings are displayed, however no slicing happens for the upper part of the model. 

By default Prusa Slic3r generates only the first 25 layers @ 0.2mm, instead of cca. 69 layers required for the model.

Saving a 3mf project containing the model and reopening it in Prusa Slic3r, the problems are detected, repaired and the model is sliced correctly.
Manually repairing the STL file, and importing the repaired model also works correctly.

* _Steps needed to reproduce the problem_
Just slice the imported model
* _Expected Results_
The model should be auto-repaired, as it happens when loading the .3mf project. Slicing should happen for the whole model (cca. 73 layers).
Alternatively, error messages could be displayed regarding problems with slicing the model that need repairing, and the user could be prompted for action.
In Cura this model can be sliced without any issues/errors.

* _Actual Results_
Slicing result contains only 25 layers. No errors or warnings displayed.

**Model:**
[image]
**Slicing result (25 layers, no errors displayed):**
[image]
**Expected Result (73 layers, Auto-repaired 292 problems):**
[image]

#### Project File (.3MF) where problem occurs

[Clips_18x.zip](https://github.com/prusa3d/PrusaSlicer/files/3639726/Clips_18x.zip)

bubnikv, 2019-09-23:
Most likely the model is corrupted. You may try to fix it with netfabb
https://tools3d.azurewebsites.net/

We will look into your model one day to try to improve slicer's own model fixing heuristics to work around your model issues.

cristiansen-12, 2019-09-23:
Thanks for the reply. I also assume there is some kind of problem with the model.

Mostly I decided to open the issue as an "inconsistent behaviour", since Prusa Slic3r in some situations correctly notifies of issues with the model and repairs them (like in the case of .3mf projects), and sometimes it just fails to work properly without any hint of a problem  (the case of the .stl model import).

Jebtrix, 2019-09-23:
Strange, that mesh is clean actually. Meshlab and blender detects nothing wrong.

Jebtrix, 2019-09-23:
Merging dup vertices does fix the import in PrusaSlicer. Didn't think PrusaSlicer had a problem with that. In case you want to look into it more @bubnikv 
[Clips_18x_merge_dup_vertices.zip](https://github.com/prusa3d/PrusaSlicer/files/3642918/Clips_18x_merge_dup_vertices.zip)

bubnikv, 2019-09-24:
@jetbrix thanks, we will look at the stl import issues in one wave.

rtyr, 2021-06-09:
It's sliced correctly on current master. It was likely fixed with 9c5dd35 (attached STL was created in OpenSCAD). Closing.

closed 2021-06-09 by rtyr