prusa3d/PrusaSlicer #2977: [Regression] Segfaulting on deleting any objects from bed

Jebtrix, 2019-09-22:
### Version
version_git-2.1.0-r139-g927ad5776
wxWidgets v3.0.4
wxWidgets v3.1.2

### Operating system type + version
Arch Linux
* -DSLIC3R_STATIC=0 
* -DSLIC3R_FHS=1 
* -DSLIC3R_WX_STABLE=0 | 1
* -DSLIC3R_GTK=3 
* -DSLIC3R_PCH=0 

### Behavior

Latest git is segfaulting on deleting any object. 2.1.0 release was fine.

```
#0  0x0000555555cf95e3 in Slic3r::GUI::ObjectList::list_manipulation(bool) ()
#1  0x0000555555cf9acd in  ()
#2  0x00007ffff7cfb89e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#3  0x00007ffff7cfbc1b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#4  0x00007ffff7cfbcb1 in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#5  0x00007ffff7cfbd64 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#6  0x00007ffff7cfbe02 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#7  0x00007ffff7cfbba7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#8  0x00007ffff74687f1 in  () at /usr/lib/libwx_gtk3u_adv-3.0.so.0
#9  0x00007ffff5aa3b4a in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#10 0x00007ffff5aa46bb in g_signal_emit_by_name () at /usr/lib/libgobject-2.0.so.0
#11 0x00007ffff5ed9e8a in  () at /usr/lib/libgtk-3.so.0
#12 0x00007ffff5ab1d3a in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#13 0x00007ffff5a9f88e in  () at /usr/lib/libgobject-2.0.so.0
#14 0x00007ffff5aa398a in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#15 0x00007ffff5aa47f0 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#16 0x00007ffff745f65c in  () at /usr/lib/libwx_gtk3u_adv-3.0.so.0
#17 0x00007ffff73bb470 in wxDataViewModel::ItemDeleted(wxDataViewItem const&, wxDataViewItem const&) () at /usr/lib/libwx_gtk3u_adv-3.0.so.0
#18 0x0000555555d88261 in ObjectDataViewModel::Delete(wxDataViewItem const&) ()
#19 0x0000555555cecc3d in Slic3r::GUI::ObjectList::delete_object_from_list(unsigned long) ()
#20 0x0000555555cece0f in Slic3r::GUI::ObjectList::delete_from_model_and_list(ItemType, int, int) ()
#21 0x0000555555e9f02a in Slic3r::GUI::Selection::erase() ()
#22 0x0000555555ca3bd5 in Slic3r::GUI::Plater::remove_selected() ()
#23 0x00007ffff7cfb89e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#24 0x00007ffff7cfbc1b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#25 0x00007ffff7cfbcb1 in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#26 0x00007ffff7cfbd64 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#27 0x00007ffff7cfbe02 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#28 0x00007ffff7cfc817 in wxEvtPlace an object on the bedHandler::ProcessPendingEvents() ()
    at /usr/lib/libwx_baseu-3.0.so.0
#29 0x00007ffff7b8e9e8 in wxAppConsoleBase::ProcessPendingEvents() ()
    at /usr/lib/libwx_baseu-3.0.so.0
#30 0x00007ffff773ad7a in wxApp::DoIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#31 0x00007ffff773ae94 in  () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#32 0x00007ffff59c527f in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#33 0x00007ffff59c71c1 in  () at /usr/lib/libglib-2.0.so.0
#34 0x00007ffff59c80d3 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
--Type <RET> for more, q to quit, c to continue without paging--c
#35 0x00007ffff600253f in gtk_main () at /usr/lib/libgtk-3.so.0
#36 0x00007ffff7757c46 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#37 0x00007ffff7bc8bae in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#38 0x00007ffff7b8d517 in wxAppConsoleBase::MainLoop() () at /usr/lib/libwx_baseu-3.0.so.0
#39 0x00007ffff7c154c8 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#40 0x0000555555890741 in Slic3r::CLI::run(int, char**) ()
#41 0x000055555587f395 in main ()
```

YuSanka, 2019-09-23:
Thanks for your report. My fault :(

Jebtrix, 2019-09-23:
All good :)

closed 2019-09-23 by Jebtrix