prusa3d/PrusaSlicer #2979: Can't start new version 2.1.0, install

YMita2017, 2019-09-23:
### Version
2.1.0

_Use `About->About PrusaSlicer` for release versions_
Can't check because not start
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
???
### Operating system type + version
Windows 10 Home 64 bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
BLV mgn Cube (Handmade) with Duet2 Wifi + Reprap firmware Version 2.03

### Behavior
Using version 2.0.0 machine, install new version 2.1.0 with all default settings (nothing change) 
Version 2.0.0, * _Describe the problem_
First time start from any Icon, Windows start menu, directory "prusa-slicer.exe"/"prusa-slicer-console.exe" double click, open information dialog as "unhandled unknown exception; terminating the application", and click "OK" button, next dialog as "Internal error: Flow::spacing() produced negative spacing. Did you set some extrusion width too small?", click "OK" button, nothing work.
I re-install 2.0.0 and its work well, no problem.

_Is this a new feature request?_ No.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-09-23:
The  "Internal error: Flow::spacing()" message explains that there is an
issue with your configuration. Please either provide the content of your
config directory, or try to make a backup of it and add your profile files
one by one.

bubnikv, 2020-03-15:
stalled issue, no data provided. hopefully this issue has been solved. Closing.

closed 2020-03-15 by bubnikv

TurielD, 2020-03-21:
Issue is not solved - it still causes a n unandled exception if you set up your nozzle too small.

Example settings from Roaming\PrusaSlicer\printer\  .ini file:

`# generated by PrusaSlicer 2.1.1+win64 on 2020-03-21 at 17:48:25 UTC
bed_custom_model = 
bed_custom_texture = 
bed_shape = 0x0,68.04x0,68.04x120.96,0x120.96
before_layer_gcode = 
between_objects_gcode = 
cooling_tube_length = 5
cooling_tube_retraction = 91.5
default_filament_profile = ""
default_print_profile = 
deretract_speed = 0
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
extra_loading_move = -2
extruder_colour = ""
extruder_offset = 0x0
gcode_flavor = reprap
high_current_on_filament_swap = 0
host_type = octoprint
inherits = 
layer_gcode = 
machine_max_acceleration_e = 10000,5000
machine_max_acceleration_extruding = 1500,1250
machine_max_acceleration_retracting = 1500,1250
machine_max_acceleration_x = 9000,1000
machine_max_acceleration_y = 9000,1000
machine_max_acceleration_z = 500,200
machine_max_feedrate_e = 120,120
machine_max_feedrate_x = 500,200
machine_max_feedrate_y = 500,200
machine_max_feedrate_z = 12,12
machine_max_jerk_e = 2.5,2.5
machine_max_jerk_x = 10,10
machine_max_jerk_y = 10,10
machine_max_jerk_z = 0.2,0.4
machine_min_extruding_rate = 0,0
machine_min_travel_rate = 0,0
max_layer_height = 0
max_print_height = 200
min_layer_height = 0.07
nozzle_diameter = 0.05
parking_pos_retraction = 92
print_host = 
printer_model = 
printer_notes = 
printer_settings_id = 
printer_technology = FFF
printer_variant = 
printer_vendor = 
printhost_apikey = 
printhost_cafile = 
remaining_times = 0
retract_before_travel = 2
retract_before_wipe = 0%
retract_layer_change = 0
retract_length = 2
retract_length_toolchange = 10
retract_lift = 0
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
serial_port = 
serial_speed = 250000
silent_mode = 1
single_extruder_multi_material = 0
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
thumbnails = 
toolchange_gcode = 
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
variable_layer_height = 1
wipe = 0
z_offset = 0
`

bubnikv, 2020-03-21:
Please test with PrusaSlicer 2.2.0-rc5, to be released as final any time soon.