prusa3d/PrusaSlicer #2982: Using an infill modifier object results in gap between perimeters

geeksville, 2019-09-23:
Hi,

I'm a fairly experienced Prusa user (and embedded firmware dev).  I was just using the modifier feature in recent PrusaSlicers and noticed a bug:

If you use a modifier (in my case three cylinders generated in my cad package) to force infill to be 100% it seems to generate a gap where that modifier's perimeter overlaps with the modified object.  This gap continues from the outer layers all the way through to the infill.  This seems to be almost exactly what you would not want, because if you are adding 100% infill to a region, you  presumably want it strongly connected to the rest of the part.

See attached photo or either of the two objects in the attached 3mf file.  I'm happy to help if you want more info.

[image]

[image]

[image]

### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer-2.1.0+linux64-201909160925.AppImage (release version)

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 18.04 

### 3D printer brand / version + firmware version (if known)
Prusa mk3s.  Pretty recent firmware - but bug can be seen with gcode viewer - no need to print.

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
[project-3mf-file.zip](https://github.com/prusa3d/PrusaSlicer/files/3643528/new.bike.bracket.zip)

Jebtrix, 2019-09-23:
Upstream Slic3r definitely has better results. It would also be useful to be able control modifier infill to infill interface/overlap percentage for fine tuning.
[image]

bubnikv, 2019-09-24:
AFAIK the upstream slic3r changed at one point of time, when the "infill /
perimeter overlap" is applied. It looks like the overlap is applied to the
transitions of the infill regions.

The good news is, that the PrusaSlicer master now merges the top regions
back if they were split by the modifier mesh but the modifer parameters do
not influence the top infill parameters.

neophyl, 2019-09-24:
Will that also work so it doesnt effect the bottom layer fill too ?

geeksville, 2019-09-24:
Yes, my photo was of the bottom, which still seems to be busted.  Also I could see as it was printing the infill the rest of the infill was never connecting to the region that had the modifier mesh - so that seems busted also.

Jebtrix, 2019-09-24:
To be fair with Gyroid infill even with upstream Slic3r doesn't exactly overlap the infill to infill interface.

[image]

witorwitor, 2019-11-08:
the gap in the surfaces that should be solid really makes this modifier meshes almost unusable If the layer is already solid, It should just be left solid.

sp847242, 2020-01-30:
I just gave 2.2.0 alpha 2 a try. This is still happening. I was trying to use a modifier to eliminate some tiny  gap-fills, but leave the longer ones, by setting Gap Fill speed to 0 within the modifier. That much worked, but the perimeters still get diced up around the modifier.

foreachthing, 2020-01-30:
If it's just for infill: place you modifier above all of your bottom layers.
That way it won't interfere with the solid layers.

sp847242, 2020-01-30:
I did see that some adjustments to the infill modifier behavior in v2.2.0, like when the modifier volume is entire within the part. But a lot of things I print are thin, and mostly or entirely perimeters. If I disable the Detect Thin Walls option, sometimes the prints get excessive material where the inner perimeters are very close. 

Images: A representative part I drew up. The long gap-fill around the main perimeter is useful and I want to keep it. But that little "point" of gap-fill inside the part is something I can do without.
So I added a Modifier cylinder, and changed the gap-fill speed to zero. All other settings are the same.
The gap-fill is indeed gone inside the main curved perimeter, but now I've got perimeters outlining the modifier.

[image]
[image]

bubnikv, 2020-02-09:
The original issue (unecessary split of the top surface, sparse gyroid infill not connected to the dense infill produced by a modifier mesh) has been resolved in PrusaSlicer 2.2.0-alpha4.

Box with a hole and a modifier around the hole increasing infill density of the gyroid infill from 25% to 40%
[image]

[image]

[image]

I am closing this issue. Please open a new issue for a topic, which is different from the original issue.

closed 2020-02-09 by bubnikv

sp847242, 2020-02-10:
Bubnikv,
I just tried this out with 2.2.0-alpha4. Same issue.
It's also got some odd perimeter stubs inside the breaks.

[image]
[image]
[image]

lukasmatena, 2020-02-10:
@sp847242 
> I think there are two separate issues being mixed in together: One deals with the infill being broken apart, the other deals with the perimeters.

Well, in fact it was you who mixed them. The original issue was indeed about unnecessary splitting of top/bottom infill and it was resolved in 2.2.0-alpha4, so it should be closed and
> a new issue for a topic, which is different from the original issue.

shall be opened.

sp847242, 2020-02-10:
@lukasmatena Ok, apologies, I see it now: I was looking at the external disruption to the pictured printed parts as being perimeter, when it was in fact external rectilinear infill.

Should I open a new thread, or is that something you've done already?

lukasmatena, 2020-02-11:
@sp847242 I have not opened anything, but I think it would be a duplicate of  #3370 and #3495. Actually, it is probably not necessary to file a new issue, this is tracked already.