prusa3d/PrusaSlicer #2983: “PrusaSlicer.app” can’t be opened because it is from an unidentified developer.

cassianoleal, 2019-09-23:
### Version
2.1.0

### Operating system type + version
macOS (any version)

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
* When opening PrusaSlicer, macOS tells me that `“PrusaSlicer.app” can’t be opened because it is from an unidentified developer.`.

* _Steps needed to reproduce the problem_
  * Open `.dmg` file, drag app to `Applications`, open it.

* _Expected Results_
* PrusaSlicer opens

_Is this a new feature request?_
No, this is a security bug.

Signing gives users a much better experience and sense of security since then we can trust that the file that's just been downloaded has not been tampered with by a third party.

This had already been the case in previous versions (#2269). Makes me wonder how broken your build pipelines might be. 😞

fiveangle, 2019-09-24:
Cannot reproduce.  Worked as expected on 10.14.6 without issue.

JohnOCFII, 2019-09-24:
Works fine here on 10.14.6 as well.

bubnikv, 2019-09-24:
@cassianoleal We have a report from one college who is using the latest (likely a beta) OSX, where he gets this warning. Other colleges did not report such issue. I suppose it is a quirk of the latest OSX beta.

cassianoleal, 2019-09-24:
Ah you're right about the fact that this is on the latest Catalina Beta, and sorry for not mentioning it. If people in previous versions of macOS are not seeing the warning, this hints to the app not having been [notarised by Apple](https://developer.apple.com/documentation/security/notarizing_your_app_before_distribution).

This is not a _quirk of the beta_. It's a feature of macOS starting with Catalina. It will not go away after the beta.

[More info on notarisation](https://appletoolbox.com/everything-you-need-to-know-about-app-notarization-in-macos-catalina/).

fiveangle, 2019-09-27:
Sounds like an invalid bug until the GM as there’s been a lot of backlash on it. Apple may capitulate and delay rolling the requirements out until later.

cassianoleal, 2019-09-28:
@fiveangle Do you have any links to where you got this information? The latest bit of official news from Apple seems to be [this piece](https://developer.apple.com/news/?id=09032019a) from September 3rd which simply states the notarisation requirements have been relaxed and how.

PrusaSlicer 2.1.0 has not been notarised regardless, and gives the warning when it's first launched.

The current beta has not been updated in many weeks and in general Apple don't add or remove much in terms of functionality and features just before launch and without a lot of testing.

Also, may I ask why you thing this is an invalid bug? I feel that something that will make the software break in a predictable future should be addressed before things go wrong, not after.

cassianoleal, 2019-10-08:
Now that [Catalina is out](https://www.apple.com/newsroom/2019/10/macos-catalina-is-available-today/) and notarisation is still very much a thing, can this be considered an actual bug on a production product?

vojtechkral, 2019-10-08:
@cassianoleal Yes. This issue is open.

cassianoleal, 2019-12-13:
Fixed in [2.1.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.1.1).

Thanks to the PrusaSlicer team!

closed 2019-12-13 by cassianoleal