prusa3d/PrusaSlicer #2984: Circle slicing issue

cpefrank, 2019-09-23:
### Version
2.1.0
Rc

### Operating system type + version
Windows 7 64bit

### 3D printer brand / version + firmware version (if known)
Tronxy x5s

### Behavior
When the slicer implements circles, such as the top smokestack on 3Dbenchy.slt, it doesn't do a reguy circle. What ends up happening is the printer smushes the top part over and over doing a very jerky back and forth motion from left to right. It's quite violent. I tested with a larger circle and it did something similar. It did the outer perimeter fine but ended up doing the violent jerky back and forth motion inner layers. I have never seen this issue, cura prints circles fine.
Thanks

Jebtrix, 2019-09-23:
It's hard to help without seeing an attached zipped 3MF project to review. It could be gap fill or with your settings/printer it has enough room for infill.

bubnikv, 2020-03-15:
no data, closing.

closed 2020-03-15 by bubnikv