prusa3d/PrusaSlicer #2985: Wrong Slicing at symmetric part

MasterMindBS, 2019-09-24:
### Version
PrusaSilcer 2.1.0+

### Operating system type + version
PrusaSlicer
Version:   2.1.0+
Build:     PrusaSlicer-2.1.0+-201909160907

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.14.6 (Build 18G95)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 NVIDIA-12.0.24 355.11.10.50.10.103
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 780M OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_multisample_filter_hint
GL_NV_point_sprite
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S
fully assembled model bought May 2019
Firmware: 3.8.0-2684

### Behavior
non symmetric slicing at total symmetric part.
Left model part is correct - Right model part has wrong slicing with dedicated a overfill.
Layer 5 is totally symmetric.
see picture 
[image]

Layer 10 also ok, but different between left and right.
[image]
 
Then Layer 11 creates a problem with an overfill-
[image]

Layer 12 makes it more worth and create an unuseful printing model
[image]

No change with a restarted PrusaSlicer. 
No solution after resetting all parameters to default.
No different slicing result at any other layer height (checking 15um to 30um)
No change with different resettings to different filaments

I expect for a totally symmetric part a symmetric slicing result.

This failure is just recognised by a wrong unusable part, because this right zone should be a hinge like the other side, where is printed correctly.

Original model and imported geometry shows no problem. See picture
[image]

additionally add the 3MF-File as zip

[PrusaSlicer_Failure_20190924.zip](https://github.com/prusa3d/PrusaSlicer/files/3645804/PrusaSlicer_Failure_20190924.zip)

Please send me any information, because that part is great and necessary. I will print it as a good useful 3D-model.

Thanks for any help.

Joe Schindler

neophyl, 2019-09-24:
The model isn't completely flat.  If you select the place to bed tool and look at the available selection areas you will see that as far as Prusa Slicer can tell the bottom isn't flat/symmetrical.  When you separated the parts they all individually place flat to the bed and so when sliced it looks symmetrical afterwards.

MasterMindBS, 2019-09-24:
Thanks for the first feedback, but as I imported once more the model. I don't see any NON FLAT surface.

See picture below back side:
[image]

and top side (see in second screenshot
[image]

That the topside is not flat is clear to me.

Attached find the "place to bed" - View from bottom in Model view
[image]
 
Showing flatness in slicing view - with constant touch on the bed: - View from bottom side in Slice Mode
[image]

Detaillied View to the incorrect layer - View from bottom side in Slice Mode
[image]

This above explanation doesn't help and don't clarify the wrong slicing topic. Sorry.
Please just try it by yourself. 

By reading the the comments at the thingiverse part. You find the same comment. Not sure if the user use PrusaSlicer.

Any new idea? Or is this a bug in PrusaSlicer?

I'm happy to support.

neophyl, 2019-09-24:
First of all your zip file when extracted on a windows pc results not in a .3mf file but a collection of files and an .amf model (no stl in sight).

When I imported it earlier at work the imported .amf file was showing the bottom place to bed area as having a triangular area missing from the bottom 'shrink wrap' effect and when selected as the area to place it put the entire model at a strange non flat angle depending on where abouts I clicked.  Wish Id taken a screen shot now as when I load it in on my home computer that I am currently on its showing as flat as in your screen shots.

Then again now that it is importing with the flat area on the bottom its also slicing correctly for me too.
[image]

The only other thing I noticed different between the images you posted and mine is that when I import the .amf file from your zip archive the model has 160 auto repaired errors.  This is something that Prusa Slicer does automatically if installed on Windows 10.  My work PC is windows 7 which may explain why it loads different for me at home and as I see you are on a Mac it wont auto repair the model for you either.  

Have you tried sending the model to Netfabb to do a repair ?  I suspect there is something broken with it which is why it isnt slicing properly for you.  Broken models are VERY common on thingiverse and most slicers usually have a go at repairing them so many will still print.

MasterMindBS, 2019-09-24:
Hi neophyl,

many thanks for your support. Yes I can only review the model with Mac and PrusaSlicer for Mac Os.

See my first import - No failure in the geometry
[image]

As my experience PrusaSlicer already repair any geometry wich has defects just in the Import Operation. So no problem is indicated or not shown.

Then with your input I repair the STL 3D-model within PrusaSlicer by File -> STL-Repairing.
A new 3D-model created as an OBJ-File.

After reload the new repaired OBJ-File I recognize the 160 failures to describe in your threat.
See here.
[image]

By re-slicing this part. Everything works fine. See in picture. No problem at the bad area.

BUT
If I import a 3D-model in PrusaSlicer, my expectation is that failures will be MARKED, so I can review an issue or the PrusaSlicer 2.1.0 needs to fix it by its own, because I can´t identify the "bad" STL-file. 

Again Many thanks for your support.

I'm still waiting for a official statement from the Prusa R&D Team. 
I will support everybody and do almost anything to generate a good software for "normal" users to support the 3D-Printing idea, process and business.

Jebtrix, 2019-09-24:
@bubnikv Another model solved by just merging dup vertices like #2974 

[FlipPhoneHolder_merged_vertices.zip](https://github.com/prusa3d/PrusaSlicer/files/3648827/FlipPhoneHolder_merged_vertices.zip)

FidelCapo, 2021-12-02:
This issue should be fixed.
Closing.

closed 2021-12-02 by FidelCapo