prusa3d/PrusaSlicer #2986: Adding Supports limit print speed to 24mm/s

garv3, 2019-09-24:
### Version
2.1.0 win64

### Operating system type + version
Windows 10 Pro 1903

### Behavior
* _Describe the problem_
Whenever I add supports to my print, the auto speed is limited to 24mm/s for the whole print.
* _Steps needed to reproduce the problem_
Add supports with auto speed enabled in the print settings, slice and check the print speeds.
* _Expected Results_
The print speeds should not be affected at all.
* _Actual Results_
The max. print speed is 24mm/s.

Without supports (see print time and speed):
[image]

With supports (see print time and speed):
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Plater.zip](https://github.com/prusa3d/PrusaSlicer/files/3647806/Plater.zip)

Jebtrix, 2019-09-24:
Your right it does. If there is any support layers they become part of the calculation to find target volumetric speed for _autospeed_. Its basically trying to find a one size fits all target for the entire print. 

> In order to honor max_print_speed we need to find a target volumetric speed that we can use throughout the print. So we define this target volumetric speed as the volumetric speed produced by printing the smallest cross-section at the maximum speed: any larger cross-section will need slower feedrates.

Support layers becomes the lowest common denominator dragging that one size fits all target down. It seems intentional. To workaround the issue setting the Support Material speed to any value works. You could use the roughly 15mm/s its setting with Auto (if it works fine for you) and set it manually.

garv3, 2019-09-24:
Oh, that's a shame. I thought auto speed would set individual speeds for all parts of the print.

Jebtrix, 2019-09-24:
Just look at the default Prusa printer profiles. For best prints it really does take individually tweaked values for most settings. Only test-n-tune and trial-n-error will give best results. Anet doesn't supply Slic3r/PrusaSlicer profiles so you will have to do some research and reference their Cura profiles to build a proper profile yourself.

ACETyr, 2019-10-06:
I'd call it a bug non the less - Volumetric speed as a feature works great and is there for a reason. Killing it with enabling supports can hardly be intentional.

mdealer, 2020-03-13:
Just as I thought I'd try this feature with a print that takes too long to tune, I find out the feature is broken.

rtyr, 2021-02-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/984. It should be fixed in upcoming releases. Closing.

closed 2021-02-03 by rtyr