prusa3d/PrusaSlicer #2987: Frozen Mac trackpad by checking the slicing layers

MasterMindBS, 2019-09-24:
### Version
PrusaSlicer 2.1.0+ 

### Operating system type + version
PrusaSlicer
Version:   2.1.0+
Build:     PrusaSlicer-2.1.0+-201909160907

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.14.6 (Build 18G95)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 NVIDIA-12.0.24 355.11.10.50.10.103
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 780M OpenGL Engine
GLSL version: 1.20

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S
fully assembled model bought May 2019
Firmware: 3.8.0-2684

### Behavior

To check and verify the slicing layers in the Layer Mode, selecting the top layer and scroll down the mac trackpad froze.

No additional "press bottom" is possible. The trackpad not react to any press.

To solve the issue I need to lift the trackpad from the table.
Then the trackpad works ok until a next "Press Bottom" is needed in the Slicing layer window.

Never happened outside the "Slicing Mode" and never happened at any other software program.

I think, that failure is just come up with PrusaSlicer. Never happened with Slicr.
Same issue since the last two PrusaSlicer updated I made.

Need help to solve the stupid topic, because that interrupt a continuous workflow.

This is a bug in PrusaSlicer!

FidelCapo, 2019-09-25:
I'm sorry, but I probably didn't understand your problem. Can you please attach a video that explains it more? Thank you.

MasterMindBS, 2019-09-25:
Hi FidelCapo,

I be not sure if a video helps.
I have a mac with the original trackpad. For any navigation I need to press on the trackpad and move the pressed finger around. So I can navigate any view with a different angle to inspect the 3D model.
To keep the view, I uplift the finger and the position of the view point is fix. 

Then sometime the uplift of the finger get no reaction of the trackpad (seems to be frozen at "Press Bottom" and the view and the navigation is still in the cursor.

It is not a really re-produceable situation. So for today I can not made any video.
Only notice is a frozen "press Bottom" at the trackpad.

FidelCapo, 2019-09-25:
So can it be a HW issue with your trackpad? I have MacBook Pro with trackpad and I never notice similar issue.

MasterMindBS, 2019-09-25:
Hi FidelCapo,

I never had this failure at any other application. Because I user a wide range of software from Office-Product to Photoshop and Garageband. I never recognise this effect. The only program ist PrusaSlicer.
As I remember at Slicr (previous PrusaSlicer) I didn't had the issue.
So from my view a hardware issue is excluded.

Any other ideas?

bubnikv, 2019-09-26:
> To solve the issue I need to lift the trackpad from the table.

Are you saying you need to lift the whole trackpad device from the table?

What is the exact model of your trackpad? I am afraid that we will not be able to help you until we get such a device connected to our computer and see the issue with our own eyes.

Is anybody else experiencing this issue?

MasterMindBS, 2019-09-29:
Dear Bubnikv,
YES, you understood this right. I have to lift up the physical original Apple trackpad from the table, then afterwards is works normally until it happened again.

The Trackpad model is Model No.A1339.

MasterMindBS, 2021-01-03:
Up to know didn't receive that problem any more. Change the Apple Mac Trackpad to the new version.
Had this problem at non other software as PrusaSlicer.

closed 2021-01-03 by MasterMindBS