prusa3d/PrusaSlicer #2988: Slicer crashes when slicing for flexible filament

fluksh, 2019-09-24:
### Version
version 2.1.0+win64, Release version

### Operating system type + version
Win 10 64 bit, latest version

### 3D printer brand / version + firmware version (if known)
MK3s stock, firmware v3.8.0

### Behavior
* PrusaSlicer crashes when slicing

Steps: 
* Open M6hexkappe.m3f file
* click slice

#### Project File (.3MF) where problem occurs

[M6hexkappe.zip](https://github.com/prusa3d/PrusaSlicer/files/3649263/M6hexkappe.zip)

fluksh, 2019-09-24:
same .m3f loads and slices correctly in version  2.1.0-alpha1+win64-201908091120

lukasmatena, 2019-09-24:
@bubnikv
I can confirm the crash. It seems to have been broken between 2.1.0-beta3 and 2.1.0-rc by commit 206dbb8.

The crash happens in GLCanvas3D.cpp:2176 at
`if (! m_volumes.volumes[idx_volume_src]->print_zs.empty())` with idx_volume_src=9 and m_volumes.volumes.size()=9. Full backtrace below.

<details>
        <summary>Backtrace from ASAN</summary>
=================================================================
==7844==ERROR: AddressSanitizer: heap-use-after-free on address 0x6190005ec308 at pc 0x000002a09f5f bp 0x7ffc72bfc280 sp 0x7ffc72bfc270
READ of size 8 at 0x6190005ec308 thread T0

    #0 0x2a09f5e in Slic3r::GUI::GLCanvas3D::load_gcode_preview(Slic3r::GCodePreviewData const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2a09f5e)
    #1 0x2cdcc6a in Slic3r::GUI::Preview::load_print_as_fff(bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2cdcc6a)
    #2 0x217d4e5 in Slic3r::GUI::Plater::priv::update_fff_scene() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x217d4e5)
    #3 0x21d07e7 in Slic3r::GUI::Plater::priv::on_process_completed(wxCommandEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x21d07e7)
    #4 0x313321d in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1396
    #5 0x313321d in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1371
    #6 0x31339b5 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) src/common/event.cpp:1868
    #7 0x3133b1f in wxEvtHandler::TryHereOnly(wxEvent&) src/common/event.cpp:1589
    #8 0x3133bc6 in wxEvtHandler::TryBeforeAndHere(wxEvent&) include/wx/event.h:3873
    #9 0x3133bc6 in wxEvtHandler::ProcessEventLocally(wxEvent&) src/common/event.cpp:1526
    #10 0x3133c70 in wxEvtHandler::ProcessEvent(wxEvent&) src/common/event.cpp:1499
    #11 0x31330b8 in wxEvtHandler::ProcessPendingEvents() src/common/event.cpp:1363
    #12 0x304820e in wxAppConsoleBase::ProcessPendingEvents() src/common/appbase.cpp:566
    #13 0x304820e in wxAppConsoleBase::ProcessPendingEvents() src/common/appbase.cpp:545
    #14 0x2e82334 in wxApp::DoIdle() src/gtk/app.cpp:150
    #15 0x2e8240f in wxapp_idle_callback src/gtk/app.cpp:100
    #16 0x7f2dce3af049 in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a049)
    #17 0x7f2dce3af3ef  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a3ef)
    #18 0x7f2dce3af711 in g_main_loop_run (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a711)
    #19 0x7f2dcfbdb696 in gtk_main (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x130696)
    #20 0x2fdb2e4 in wxGUIEventLoop::DoRun() src/gtk/evtloop.cpp:65
    #21 0x3073441 in wxEventLoopBase::Run() src/common/evtloopcmn.cpp:76
    #22 0x3048353 in wxAppConsoleBase::MainLoop() src/common/appbase.cpp:380
    #23 0x30b879e in wxEntry(int&, wchar_t**) src/common/init.cpp:507
    #24 0xa2ef32 in Slic3r::CLI::run(int, char**) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xa2ef32)
    #25 0x9e07e2 in main (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x9e07e2)
    #26 0x7f2dcd21c82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #27 0xa19f48 in _start (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xa19f48)

0x6190005ec308 is located 904 bytes inside of 944-byte region [0x6190005ebf80,0x6190005ec330)
freed by thread T0 here:

    #0 0x7f2dd1212caf in operator delete(void*) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10ecaf)
    #1 0x29f0cb7 in Slic3r::GUI::GLCanvas3D::reset_volumes() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x29f0cb7)
    #2 0x2ce7a1e in Slic3r::GUI::Preview::reload_print(bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2ce7a1e)
    #3 0x217d4e5 in Slic3r::GUI::Plater::priv::update_fff_scene() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x217d4e5)
    #4 0x21d07e7 in Slic3r::GUI::Plater::priv::on_process_completed(wxCommandEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x21d07e7)
    #5 0x313321d in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1396
    #6 0x313321d in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1371

previously allocated by thread T26 here:

    #0 0x7f2dd1211d2f in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10dd2f)
    #1 0x296a377 in Slic3r::GUI::GLCanvas3D::_load_wipe_tower_toolpaths(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)::{lambda(float const*)#1}::operator()(float const*) const [clone .isra.0] (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x296a377)
    #2 0x298b00e in Slic3r::GUI::GLCanvas3D::_load_print_object_toolpaths(Slic3r::PrintObject const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<double, std::allocator<double> > const&)::{lambda(tbb::block [... cut ...] ring<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<double, std::allocator<double> > const&)::{lambda(tbb::blocked_range<unsigned long> const&)#3}, tbb::auto_partitioner const>::execute() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x298cdc6)
    #4 0x372aa94 in tbb::internal::custom_scheduler<tbb::internal::IntelSchedulerTraits>::local_wait_for_all(tbb::task&, tbb::task*) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x372aa94)

</details>

fluksh, 2019-09-24:
You're right, it is the STL causing the crash, running it through netfabb gets rid of the crash. I was rushing it in Fusion360.

Jebtrix, 2019-09-25:
There is something odd here but its not the mesh. It's the perimeter setting of 5 with the tiny area size left over for infill from the dimension of the object itself. 

* Scale it 1%  bigger or smaller it will slice with settings in that project.
* Set infill to 0% - Slices fine

Here is the project where I just deleted his mesh, right clicked and added a cylinder shape primitive, set the dimensions to 4.5mm X/Y and I get the same crash behavior.

[M6hexkappe_cylinder_primitive.zip](https://github.com/prusa3d/PrusaSlicer/files/3651425/M6hexkappe_cylinder_primitive.zip)

Jebtrix, 2019-10-04:
Digging into this I've got it to crash using pure default Prusa profiles: printer, print, and filament. All I did was added a cylinder shape, set the dimensions to 4.5mm X/Y, and change Perimeter to 5.

* 0.20mm SPEED MK3
* Generic Flex
* Original Prusa i3 MK3S

What is strange is two settings combined do this. Setting perimeter to 5 with other filaments wont crash it, Generic Flex has Filament Overrides > Length to **0** which the is required to crash. This setting was present in @fluksh original project.

[Shape-Cylinder_Defaults.zip](https://github.com/prusa3d/PrusaSlicer/files/3689405/Shape-Cylinder_Defaults.zip)

bubnikv, 2019-11-22:
Fixed with 6fcd51d5be79783037414f9ea57bbe190b82ba6c
Closing

closed 2019-11-22 by bubnikv

Drachenkaetzchen, 2019-11-23:
Same issue here. Workaround: Use a very small value (like 0.001) which doesn't crash and the FW should hopefully ignore that.