prusa3d/PrusaSlicer #2989: support_material_extrusion_width=0.35 mm is too low to be printable at a layer height 0.35 mm

WhiteDragonD, 2019-09-24:
In version 2.1.0 of Prusa Slicer, if I choose the 0.35mm Print Setting then selected Supports Everywhere, I get an error that pops up.

support_material_extrusion_width=0.35 mm is too low to be printable at a layer height 0.35 mm

I downgraded to 2.0 and this problem does not happen.

### Version
_Version of PrusaSlicer used goes here_
2.1.0

### Operating system type + version
Windows 7/10

### 3D printer brand / version + firmware version (if known)
MK2.5

Jebtrix, 2019-09-25:
Extrusion width validations was added. The reason is that creates a theoretically round extrusion that would have adhesion problems. Hence the message. Not everybody agrees with the limits but it is what it is for now. Just make it 35.01 to nudge it.

WhiteDragonD, 2019-09-25:
> 
> 
> Extrusion width validations was added. The reason is that creates a theoretically round extrusion that would have adhesion problems. Hence the message. Not everybody agrees with the limits but it is what it is for now. Just make it 35.01 to nudge it.

Question for stupid me, what/where is the setting for that?

lukasmatena, 2019-09-25:
Print Settings -> Advanced -> Extrusion width (you have to be in Advanced or Expert mode)

WhiteDragonD, 2019-09-25:
Ok how the program is functioning right now is dumb.  If I don't have supports selected on my Plater (aka set to none) the Support Material box is disabled... AND there's no message telling me why AND the hover tip is not showing.

When I change to Support Everywhere on the Plater, then the message pops up with the error without information on what/how to fix it.  If this new build is enforcing those values then some additional text would be nice to explain why and how to fix it.

After I change it to Support Everywhere, then I can change the Support Material width to 0.351 to avoid the error.

Should this be set to 0 now for better prints?

I don't change any values except for Infill, perimeters, brim, and speed, so me using the program in it's basic function, it's "broken".

lukasmatena, 2019-09-25:
> I don't change any values except for Infill, perimeters, brim, and speed

This is actually a good point. Our own MK2 profiles set both layer height and support extrusion width to 0.35mm, which is not allowed and triggers the error as soon as supports are enabled. If it appeared after someone messes with the layer height or extrusion width, the message text would be enough, but it should not happen with default profiles in my opinion.

What do you think, @rtyr?

lukasmatena, 2019-09-25:
> If I don't have supports selected on my Plater (aka set to none) the Support Material box is disabled

I personally agree that disabling the supports should not deactivate the Support Extrusion width textbox.

It's not even consistent with the other fields, if I e.g. set infill to 0%, the respective extrusion width is not disabled. By the way the extruder settings in Multiple Extruder is, and it stays disabled even if I override the infill density in the object settings.

rtyr, 2019-09-25:
I will increase support material extrusion width slightly for the "0.35mm FAST" print profile to avoid this error message.

rtyr, 2020-03-15:
Print profile was updated. Closing.

closed 2020-03-15 by rtyr

Recrera, 2021-01-19:
This is the message im getting: `support_material_extrusion_width=0.36 mm is too low to be printable at a layer height 0.36 mm`
So what's the minimum size for it?