prusa3d/PrusaSlicer #2990: Save Purging Volumes to config

jimmy00784, 2019-09-24:
### Version
Prusa Slicer 2.1.0

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Custom multi material single nozzle CoreXY printer running Klipper.

### Behavior
Purging Volumes are not saved to config when Prusa Slicer is exited. Saving purge volume matrix and loading it from config at startup would help when working with certain filaments where default purge volume doesn't result in clean prints.

_Is this a new feature request?_ Yes

lukasmatena, 2019-09-25:
The volumes are saved into a 3MF/AMF. It is not simple to decide where in the config it should be saved, because it depends on pairs of selected filaments (and their color) which can be chosen quite independently on Print/Printer Settings and each other. Also discussed in https://github.com/prusa3d/PrusaSlicer/issues/1031.

jimmy00784, 2019-09-25:
@lukasmatena While saving project files to take a snapshot of configuration is an acceptable stopgap measure, this really needs to be handled at the configuration level. An ideal implementation would handle it at a filament configuration level. Please note that default settings work most of the time and it is only for certain filament pairs that this setting is ever needed to deviate from defaults.

See my reference screenshot below for instance. Transitioning to Tianse Black PLA or from Tianse White PLA only requires a small amount of filament purge, but transitioning from Tianse Black PLA or to Tianse White PLA requires large purge and that might also vary depending on what filament you might transition to. Moreover, I have another brand of White PLA where the numbers vary further and settings for Tianse While PLA might not be adequate.
 
[image]

At the minimum, at least allow the purge volume matrix to be saved to Printer Configuraiton so it can persist between applicaiton restarts.

Also, if this feature seems interesting enough, I have thoughts on UI elements as well as configuration that I might be able to help with.

gdombiak, 2020-01-03:
It would be helpful to remember last entered values. It's now defaulting to 70/70 and it would save me some time (and possibly a mistake) if last entered values are kept.

dintid, 2020-01-18:
I sure wish last used values were saved as well.. It's a huge bother having to enter everything every single time from some note that tends to disappear between prints.
This would be especially usefull when setting up new machines, both printers and MMUs and dialing in new filament brands/colors as it's super hard to keep track of.
Alternatively make a checkmark option where you can choose wheter or not to keep last entered values.

Pun-e, 2020-11-10:
Me three.

Just started big print after a PC restart and was kind of shocked at how slow my purging was, and now I know why.

Even if this (forgetting every time) is forever the standard behaviour, not informing new users of this fact is asinine. I lost a lot of my calibration work because I thought it would function _like literally every other setting_

Kachidoki2807, 2020-12-15:
Hello,

I have another case, when using an IDEX printer, the default purge volume is excessive. Each time I want to start a new project, I have to manually decrease theses values (5mm³ in advanced mode is enough). I just learned to compile my own build of PrusaSlicer to reduce the default values, especially for this annoying issue. But it is not a convenient way to do...

I have the same issue with the default values for the filament settings (15mm³ by default), I need to create a custom profile for each preset setting I use. Moreover that is not really a relevant value because it only applies to a single extruder multimaterial like the MMU, but not for an IDEX nor a toolchanger. At least in this case the value is saved into the custom filament profile.

shadow7412, 2021-08-20:
Has there been any movement on this? I'd also like this to be saved generally - even just overriding the default in the application settings somewhere would be great.

I have no use for purging on unload - leaving that on accidentally would do nothing but waste filament.

bubnikv, 2021-09-06:
We will allow saving an empty project file (3MF). This shall fulfill your
requirements better, as you will be able to save the purge tower settings
together with filament selection.

shadow7412, 2021-09-06:
Would we be able to set that empty project as the startup template? If so, then that could solve the issue with the purge on unload default.

But how do we save the changes we make to that template, without including the other temporary changes (including the model import) made during a print? Can we maintain the grid for different filaments (ie, colours) that way?

I think what I'd like to see is a "default purge on (un)load" setting in the filament tab, with the ability to add overrides for when interacting with specific other filaments (which you'd add to a table, which should end up looking similar to the "advanced" purge settings). After that default has been applied, it could then be overridden by the current screen.

Do you think that would work?

yonitjio, 2023-05-30:
To put purging volumes on filament's config will grow data each time a new filament added, 2x2, 3x3, 4x4, and so on. Perhaps this is why the developers are reluctant to add this.

I think we should just meet half way, just add it to the printer setting and instead of calling it filaments purging, call it extruder load and unload or something.

It still will grow exponentially each time an extruder added, but I think most of us very rarely, if ever, add a new extruder. 

Even when we do, it's in low numbers, except perhaps a virtual extruder, but again that's a very special/specific condition.

Imburr, 2023-06-01:
SuperSlicer supports this via advanced pigment algorithm. It requires saving each color/filament as its own profile, but once it is setup no more need to tune purge or wipe tower volumes. 

[image]
[image]
[image]

yonitjio, 2023-06-10:
How does it work? How do you determine values for the pigment?

Imburr, 2023-06-11:
> How does it work? How do you determine values for the pigment?

You calibrate. I typically start with my eyeballs, lighter colors are lower, darker are higher. Reds bleed a lot, so reds are higher. Then i start with high volume, and lower until i see bleed, then go back up. Then as i orint with MMU as i see bleed, i adjust pigment up or down 0.02 ish.

yonitjio, 2023-06-14:
Right, I did see the pigment but didn't know the advance calculation also must be enabled. Thank you for sharing this.

Imburr, 2023-06-14:
> Right, I did see the pigment but didn't know the advance calculation also must be enabled. Thank you for sharing this.

If you look around in the enraged carrot feeder GitHub, there is a purge algorithm spreadsheet which has the different advanced settings along with their purging volume.

audiorazor, 2023-10-17:
+1.   Would love to see something like this.   MMU doesnt need to be so much work to setup each time.

lukasmatena, 2024-03-21:
[PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1) introduced a possibility to set the purging volumes in Printer / Filament Settings. Closing.

closed 2024-03-21 by lukasmatena